winder/Universal-G-Code-Sender #210: Settings not being saved

Anachronicman, 2015-01-07:
I am having an unusual problem where the settings are not being saved. This is under Windows 7

Can anyone point me to where they are saved so I can check if everything is being created correctly.

thanks.

Darbness, 2015-01-13:
I had the same problem. My work around was to manually change the value. Ie send $200=100 to the controller.

winder, 2015-02-24:
Are these GRBL settings or UGS settings which aren't being saved?

Anachronicman, 2015-02-24:
They were UGS settings.

winder, 2015-02-25:
Replace RXTX with JSSC: bd9a4781e4f83ea21f4457fbced7ce5d723d58df 017bbbf5b5e8e3fa03a6ea770fd42eae67788dfb 6931115a20c943583974067c560e7632e0844d54 2a3b1a0c502e78ba2e6e041b0b52a6a5cd16b19a

winder, 2016-02-17:
I believe this was fixed a while ago

closed 2016-02-17 by winder