winder/Universal-G-Code-Sender #211: On nightly build version 1.0.8 Dec.29

RayMJr, 2015-01-07:
When jogging using the machine control tab, in inch mode, I cannot jog at .001.  It just send a zero movement.   It seems as if it is truncating it to two decimal places, as it would when in mm mode.

Ray

winder, 2015-01-08:
You should be able to configure how many decimal places are left after truncation in the settings.

RayMJr, 2015-01-08:
I didn't see that, but I will check again and let you know.

Ray

--J. Ray Mitchell Jr.
jrmitchellj@gmail.com
(818)324-7573

The things we admire in men, kindness and generosity, openness, honesty,
understanding and feeling are the concomitants of failure in our system.
And those traits we detest, sharpness, greed, acquisitiveness, meanness,
egotism and self-interest are the traits of success. And while men admire
the quality of the first they love the produce of the second.
-John Steinbeck, novelist, Nobel laureate (1902-1968)

RayMJr, 2015-01-09:
I check the truncation level, and moved it up to 8 & down to 1, with no effect on the jog function in the machine control tab.
This was working in the 1.0.7 version.

Ray

RayMJr, 2015-01-14:
Another issue I just came across:
When you hit the return to zero button in  the machine control tab, it seems to go to the zero point for the work coordinates you are currently in, then it immediately goes to the machine zero coordinates.

Very unexpected behavior!

Ray

commit 8490b51b (2015-02-19): Change jog spinner to round 3 places. Closes #211
closed 2015-02-19 by winder