winder/Universal-G-Code-Sender #213: translations

Majestad278, 2015-01-19:
I would like to know how to translate into my language program. provided that the author me permission to change the language.

I think the opening to foreign languages is a great opportunity for this type of program.

winder, 2015-01-19:
All you need to do is add a file to the resources directory with your country code at the end and put in the translations. If you want to download MessagesBundle.properties and send me the translated file I'm happy to integrate it, otherwise just send a pull request!

https://github.com/winder/Universal-G-Code-Sender/tree/master/src/resources

Majestad278, 2015-01-21:
thanks. I work so hard to send this file.

Majestad278, 2015-02-02:
how i can test the traduction?

lazyzero, 2015-02-03:
You can add the new traslation by adding a line in the class AvailableLanguages
https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/i18n/AvailableLanguages.java
When you now run UGS you can select your translation from the settings menu.

closed 2016-01-12 by winder