winder/Universal-G-Code-Sender #214: "Error while starting filestream: null"

villamany, 2015-01-20:
Hi, thanks for the great work!.

Im getting this error on 1.0.8 (nightly) and previous versions.
Mainly at beggining but some times on middle of a work.

Im using windows 8 x86. This is the log:

[image]

[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose]<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
Skipping command #0
>>> M5
Skipping command #2
>>> G21G90
>>> G0Z11.5
ok
Skipping command #5
>>> G17
>>> G0X0.0Y0.0
>>> Z0.1
>>> G1S1.0F30000.0
ok
ok
ok
>>> S0M3
>>> Y44.0568S124.3333F1000.0
>>> X0.1292
ok
ok
ok
ok
>>> X0.2584
>>> X0.3876
>>> X0.5168
>>> X0.646
>>> X0.7752
>>> X0.9044
>>> X1.0336
>>> X1.1628
>>> X1.292S141.0
ok
>>> X1.4212S124.3333
>>> X1.5504
>>> X1.6796
ok
>>> X1.8088
ok
>>> X1.938
ok
>>> X2.0672
ok
>>> X2.1964
ok
>>> X2.3256
ok
>>> X2.4548
ok
>>> X2.584
ok
>>> X2.7132
ok
>>> X2.8424
ok
>>> X2.9716
>>> X3.1008
ok
>>> X3.23
[... 150 lines cut ...]
>>> X35.0129
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.8837
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.7545S91.0
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.6253S74.3333
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.4961
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.3669
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.2377S91.0
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X34.1085S74.3333
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X33.9793
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X33.8501
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X33.7209
[verbose][Error] Error while processing response <ok>: Attempting to complete a command that doesn't exist: <(-1)>ok
>>> X33.5917

..Continue this error....

xkobalx, 2015-01-31:
I'm getting the same issue. Here is what I was able to capture from the console.
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException:
 9436 >= 7078
        at java.util.Vector.elementAt(Unknown Source)
        at javax.swing.table.DefaultTableModel.getValueAt(Unknown Source)
        at com.willwinder.universalgcodesender.uielements.GcodeTable.updateRow(G
codeTable.java:105)
        at com.willwinder.universalgcodesender.MainWindow$59.run(MainWindow.java
:2706)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Sour
ce)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException:
 9437 >= 7079
        at java.util.Vector.elementAt(Unknown Source)
        at javax.swing.table.DefaultTableModel.getValueAt(Unknown Source)
        at com.willwinder.universalgcodesender.uielements.GcodeTable.updateRow(G
codeTable.java:105)
        at com.willwinder.universalgcodesender.MainWindow$58.run(MainWindow.java
:2684)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Sour
ce)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException:
 9437 >= 7079
        at java.util.Vector.elementAt(Unknown Source)
        at javax.swing.table.DefaultTableModel.getValueAt(Unknown Source)
        at com.willwinder.universalgcodesender.uielements.GcodeTable.updateRow(G
codeTable.java:105)
        at com.willwinder.universalgcodesender.MainWindow$59.run(MainWindow.java
:2706)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Sour
ce)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
[... 205 lines cut ...]
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException:
 9442 >= 7084
        at java.util.Vector.elementAt(Unknown Source)
        at javax.swing.table.DefaultTableModel.getValueAt(Unknown Source)
        at com.willwinder.universalgcodesender.uielements.GcodeTable.updateRow(G
codeTable.java:105)
        at com.willwinder.universalgcodesender.MainWindow$59.run(MainWindow.java
:2706)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Sour
ce)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)

jhh3000, 2015-05-16:
I just started a new issue thread called "java.lang.NullPointerException on large G-code file" but I've tried re-running my G-code file a few times and have received this exact same error. Perhaps these two are related?

Link: https://github.com/winder/Universal-G-Code-Sender/issues/256

VincentGijsen, 2015-05-31:
same issue here, a code file (rasterized image), 1385696 rows

winder, 2015-06-03:
I think there is a timing issue, I made a performance improvement in the nightly build recently which seems to exacerbate the issue -- so I'm able to reproduce the problem now.

I think somehow one of the command buffer is being exhausted before the file is done being sent. I probably could change the lists to be something like a blocking queue to resolve the problem. Its a timing problem though, so as soon as I try to set a breakpoint to see the issue happen things are magically fixed.

On my computer enabling the "scroll output window" option slows things down just enough for the timing issue to dissapear...

winder, 2015-06-10:
I believe this issue should be resolved in the current nightly build. Some of the data structures needed to be changed to thread safe versions. My guess is that this has always been an issue but it was only recently exposed due to all the performance improvements made to GRBL over the past year.

Let me know if you still see these problems.

VincentGijsen, 2015-06-10:
indeed, using a nightly build resolves big files crashing the app, nice
work!

jhh3000, 2015-06-11:
It definitely improves reliability; however I was testing a file on Grbl v 0.9i and it seems to have crashed at the very last instruction in the set (a return to zero command). It throws the error message "attempting to complete a command that doesn't exist" and gives me the last command in the file (G90 G28 X0 Y0). But the strange thing is, when I poll Grbl for the controller state, I get 0, 0, 0 which means that Grbl must have received the command...

closed 2015-07-16 by winder

klucsik, 2018-11-30:
This issue is just happend to me on my raspberi pi, i turned off the croll outut window and the UGS is working.