winder/Universal-G-Code-Sender #220: Seek to value

shmorgan, 2015-01-30:
Add a field for each of the X,Y,Z positions with a Seek/Goto button for each, This way you can type in a value, hit Go to move the head directly to a desired position.

twforeman, 2015-01-30:
This one seems silly to me. Just enter G0 X/Y/Z<value> in the command box.

shmorgan, 2015-01-30:
>  This one seems silly to me. Just enter G0 X/Y/Z in the command box.
>  -

I understand, the idea being that we do not have to KNOW Gcode, the application will take care of this for us.
 
I have a custom version I have been playing with, but I know VERY little about NetBeans and have a hard time changing even the small stuff I have so far.

twforeman, 2015-01-31:
While I appreciate the idea of "hiding the guts from the user", if you want to successfully run a CNC mill, especially a DIY one, you are probably going to need to know at least a small sub-set of G-Code.

That being said, if you think it's useful to add input boxes to the GUI interface, far be it from me to say no. It's not my project.

closed 2016-04-10 by winder