winder/Universal-G-Code-Sender #223: Error while starting file stream: null

dhughes, 2015-02-08:
I've been trying to use versions 1.0.7 through 1.0.8 nightly (as of 2/7/15). In all of these versions, but not in 1.0.6, I get frequent errors saying "Error while starting file stream: null" when trying to send work to my Shapeoko 2. I haven't found any pattern to their occurrences other than "frequently". After this error occurs I can sometimes reset my work and try again and the problem will not occur.

This problem usually presents itself very early on, typically within less than 100 lines of gcode being sent.  However, I recall seeing it happen much later on as well at least once. In some cases the machine will keep running just fine. In others it comes to a complete stop. 

Here is a screenshot of the error and the UI:

[image]

Here is the console output with the stack trace:

```
iDontKnow:UniversalGcodeSender doug$ ./start.sh
[Boot] INFO:  myJarPath=file:/Applications/UniversalGcodeSender/UniversalGcodeSender.jar
[JarClassLoader] INFO:  setProperties(com.simontuffs.onejar.JarClassLoader@7440e464)
[JarClassLoader] INFO:  oneJarPath=file:/Applications/UniversalGcodeSender/UniversalGcodeSender.jar
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:  oneJarPath=file:/Applications/UniversalGcodeSender/UniversalGcodeSender.jar
[JarClassLoader] INFO:  resources/build.properties in main/Universal G-Code Sender.jar is hidden by / (with different bytes)
[JarClassLoader] INFO:  findResource() found: "com/willwinder/universalgcodesender/Version.class" for caller null in codebase main/Universal G-Code Sender.jar
[JarClassLoader] INFO:  findResource(): unable to locate "resources/MessagesBundle_en.properties"
Feb 08, 2015 1:37:22 PM com.willwinder.universalgcodesender.SettingsFactory loadSettings
INFO: Loading settings.
Feb 08, 2015 1:37:22 PM com.willwinder.universalgcodesender.SettingsFactory loadSettings
INFO: Settings file location: /Users/doug/Library/Preferences/UniversalGcodeSender.json
Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7
java.lang.NullPointerException
	at com.willwinder.universalgcodesender.GrblCommunicator.streamCommands(GrblCommunicator.java:145)
	at com.willwinder.universalgcodesender.AbstractController.sendStringToComm(AbstractController.java:447)
	at com.willwinder.universalgcodesender.AbstractController.queueCommandForComm(AbstractController.java:436)
	at com.willwinder.universalgcodesender.AbstractController.beginStreaming(AbstractController.java:537)
	at com.willwinder.universalgcodesender.MainWindow.sendButtonActionPerformed(MainWindow.java:1677)
	at com.willwinder.universalgcodesender.MainWindow.sendButtonActionPerformed(MainWindow.java:1688)
	at com.willwinder.universalgcodesender.MainWindow.access$200(MainWindow.java:73)
	at com.willwinder.universalgcodesender.MainWindow$3.actionPerformed(MainWindow.java:286)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
[... 15358 lines cut ...]
ok
>>> X220.904Y58.875
ok
>>> X219.599Y57.455
ok
>>> X214.463Y52.055
ok
>>> X213.661Y51.178
ok
>>> X212.894Y50.310
ok
>>> X212.168Y49.450
ok
>>> X211.486Y48.595
ok
>>> X210.855Y47.744
ok
>>> X210.280Y46.895
ok
>>> X209.765Y46.047
ok
>>> X209.315Y45.198
ok
>>> X208.936Y44.346
ok

Ninjbee, 2015-02-12:
I have this error too. Usually within the first 10 seconds of sending a file. 
- Independent of whether visualiser is running.
- Only G17, G21, G0, G1 M30 and F commands used in Gcode. 
- No lines longer than 40 characters.
- Intermittent but much more common on some gcode files.
- Occurs approximately 5-10% of the time. 
- Have experienced this on two PCs - Win8 and WinXP (UGS 1.07 on both)
- Have experienced this on two Arduino uno controllers. (Same Shapeoko 2 with same grbl settings.)
- Gcode generated in Vectric Cut3d
- GRBL 0.8

Any help would be appreciated.

winder, 2015-02-24:
duplicate of #214

closed 2015-02-24 by winder

BishalSuman, 2018-02-17:
hello sir,
i am facing problem in ugs platform

here is the screen shot along with the  console output.
[image]

**** Connected to COM4 @ 115200 baud ****
Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 0    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, step/mm)
$102 = 250.000    (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)
ok
>>> $G
[verbose] GrblFeedbackMessage{message='[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]', distanceMode='G90', units='G21'}
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]
ok
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[... 56 lines cut ...]
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
[verbose] <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000>
............continuing
plz help for my college project

jahnj0584, 2018-02-17:
Your file appears to be empty...