winder/Universal-G-Code-Sender #225: java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path

natevw, 2015-02-17:
I cannot launch UniversalGcodeSender-v1.0.4-all64 on my machine (OS X 10.9.5).

```
postel:UniversalGcodeSender-v1.0.4-all64 natevw$ sh start.sh
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1119)
	at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:83)
	at com.willwinder.universalgcodesender.CommUtils.getSerialPortList(CommUtils.java:44)
	at com.willwinder.universalgcodesender.MainWindow.loadPortSelector(MainWindow.java:1133)
	at com.willwinder.universalgcodesender.MainWindow.initProgram(MainWindow.java:973)
	at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:39)
	at com.willwinder.universalgcodesender.MainWindow$26.run(MainWindow.java:967)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

The release notes for 1.0.1 include:

> New distribution jar - Application no longer requires RXTX to be installed

…and the only other mention of this dependency is that I would need to install it "for development". I am only trying to use the app, but getting this error.

winder, 2015-02-17:
Try 1.0.7 or 1.0.8, this should have been fixed a while ago.

natevw, 2015-02-17:
Ah, thanks! I think I simply clicked the "downloads page" link but I see now that is for older releases and 1.0.4 is not the current.

1.0.7 is launching for me…nice!

closed 2015-02-17 by natevw