winder/Universal-G-Code-Sender #226: Better visualizer

winder, 2015-02-20:
The replacing the yellow rectangle with a cone like chilipeppr would be nice.

The way [this page](http://nraynaud.github.io/webgcode/) has the orientation box and mini zero plane would also be nice.

winder, 2015-06-30:
Other suggestions:

Add scales along each axis.
Create buttons to go to each of the three plane views.
Add measurement feature - "how deep is the cut at this point?"
Add mm or inches to Dimensions statement.

winder, 2016-03-04:
Looks a lot nicer!

[image]

closed 2016-03-04 by winder

shmorgan, 2016-03-04:
Do you have a downloadable version of the .jar file with this?

shmorgan, 2016-03-04:
Look GREAT!!

winder, 2016-03-04:
@shmorgan It is in the nightly UGS platform build: http://bit.ly/1DkClRW

jahnj0584, 2016-03-05:
Can I assume this shows the job 0 VS where the spindle currently is? Is there a Jog To feature? I like it for making sure my material is in the right spot, when it comes to things with little spare and air cutting is boring.