winder/Universal-G-Code-Sender #229: Configurable keyboard shortcuts

winder, 2015-02-24:
Configurable binding between shortcuts and GUI operations.

goobering, 2015-03-09:
This would open up some lovely opportunities for using USB buttons to run the show, provided it's implemented as 'press the button combination for this shortcut now', as opposed to selecting a key from a list. Could I suggest implementing something to flick jog distance between 0.1/1/10 units? Whole thing externally controlled by buttons mimicking key presses would be *lovely* :)

winder, 2015-03-09:
I'd like to add a configurable key mapping for every button, this is one of the goals of a 2.0 front end refresh that I'm working on.

You can create another feature request for the 0.1/1/10 suggestion. I'm not sure I completely understand what you have in mind so a screen shot or mock up of the idea would help. One of the main goals in 2.0 is to have everything be module based so experimenting with different things like this will be much easier.

closed 2016-05-14 by winder