winder/Universal-G-Code-Sender #231: GUI Refresh - ideas

winder, 2015-02-25:
I'm starting to think about how to better organize the UGS back end so that the front end can be expanded and/or rewritten without becoming more of a mess. Most likely there will be a bunch of individual components which make up the framework rather than one monolithic class like MainWindow.java is today.

So I'm interested in what the community has for their wishlist for different modules which could be created and how they might be organized within a GUI. Keeping things in modules could make it so that we can have multiple front ends as well (i.e. separate tabless & model GUIs that share most of their code).

So here's a starting point.

* Gcode Editor with integrated visualizer (i.e. selecting a line of gcode displays in the visualizer)
* Work coordinate manager (i.e. Issue #164)

As for how the front end would be changed to incorporate these things, I'm not sure. Multiple windows like GIMP could work, but might be intimidating for new users. Tabs like the current design are good, but occasionally you might want to see multiple tabs at the same time.

rugbymauri, 2015-02-25:
Yes!

as you see on Issue  #186, i build UGS with a new GUI. Im pretty happy with it.

- I dont need a integrated visualizer, it only fills the screen with "use less" information. Normally the G-Code was alredy visulized on the CAM. Im using UGS on an ARM CPU and i uses a lot of performance.
- Yes, the main GUI-Issue for me changing the Tabs permanently.

madeinoz67, 2015-06-16:
Is there any thoughts on splitting out the backend to a GRBL Comms Library into a separate Jar with a clear API?  I'm in the process of writing a driver for Visicut for a Laser Engraver Device using GRBL and am looking look at using this to handle GRBL comms as all the hard works has been done, but don't need all the GUI.

winder, 2015-06-16:
@madeinoz67 Yes! You can do this today, take a look at [BackendAPIReadOnly.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/model/BackendAPIReadOnly.java) and [BackendAPI.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/model/BackendAPI.java). This API is used (almost) exclusively in [MainWindow.java](https://github.com/winder/Universal-G-Code-Sender/blob/fc71cf41a43120314b524ae274581641eeec1141/src/com/willwinder/universalgcodesender/MainWindow.java) and is used exclusively in the [UGSPlatform](https://github.com/winder/Universal-G-Code-Sender/tree/fc71cf41a43120314b524ae274581641eeec1141/UGSPlatform) code.

The only part that I'm not quite satisfied with is how settings are passed in. Currently there is a "Settings" object that dumps a bunch of stuff in... things like how many decimals to truncate to or whether to use single-step mode. At least the preprocessing I want to split into a separate API that you can use then use the backend API just for sending data and getting results.

madeinoz67, 2015-06-16:
> der/universalgcodesender/model/BackendAPIReadOnly.java>  and BackendAPI.java
> der/universalgcodesender/model/BackendAPI.java> . This API is used (almost)
> e274581641eeec1141/UGSPlatform>  code.
> 463201> .

Great thanks for that will investigate further.

Crazydoub, 2015-07-14:
I made some modifications to your code and put MACRO buttons ton control panel.
They automatically (de)activate while macro textbox is changed and config loaded.
It would be great to add this to future release and add keyboard link to these 5 buttons.

[image]
[image]

jahnj0584, 2016-02-22:
I like the idea of the tabless GUI, like Ruby posted (with a longer command line though, on top of the console window). Its annoying to switch em all the time without a shortcut when your mouse uses the spoilerboard as a pad :P

philreindl, 2016-03-04:
I would like to have a tab-minimal GUI, which works well on a cheap touch screen tablet.  

What I mean by tab-minimal is that the normal operation shouldn't involve switching tabs, but separate tabs for less common tasks is fine.  For example, I think it would make a lot of sense to have a page for managing macros which isn't intended to be used while running the machine.  The main screen should show the macros as non-editable buttons.  

Of course, there is probably a different set of functionality that would be needed for different "normal" operations.  Those different workflows should be self-contained in a single tab.  My opinion is that almost all of the buttons on the GUI should be essentially the same as the custom macros.  There are only a couple of special cases (file open, jog increment adjustment, jog units (Do people need to change this in normal operation?  Can jog units be moved to a setting?), etc.)  If things were set up that way, the main control area is just a bunch of macro buttons.  The set of buttons and their layout could be managed as settings rather than code.  

This is the overall thrust of the code I've been adding, just haven't had a lot of time to work on it lately.

winder, 2016-03-05:
@philreindl One idea I had early on was to integrate something like eclipse perspectives to have multiple GUI configurations in UGS platform. Some defaults I can imagine are "Sending", "Gcode Editing", "Auto Leveling", "Classic", "Tabless/Minimal", etc. Maybe even a "Tablet" mode that enables a special plugin with big buttons. NetBeans doesn't have this feature built in, but [there is a plugin](http://plugins.netbeans.org/PluginPortal/faces/PluginDetailPage.jsp?pluginid=3273) that might do the trick.

closed 2016-05-14 by winder