winder/Universal-G-Code-Sender #232: Plugin - Architecture

rugbymauri, 2015-02-25:
A big enhancement for the project will be a plungin architecture.

By putting jar's in a plugin-folder you can extend the UGS.

the Plugin-Interface shoud provide interfaces to:

- the controller, for sending G-Code directly
- ability to modify the G-Code before sending
- ability to catch the controller responses
- GUI
-- extend Standard menu
-- new Menu
-- new Tab's/Window's (depending on the new GUI)

winder, 2015-02-25:
Now thats an intriguing idea. A lot of UGS already uses observers to communicate information, and the new GUI model object is perfect for sharing the program state between plugins.

Have you done anything like this? I'll need to read up on java plugins.

Having the GUI be a plugin would open the door for things like an Android port as well.

rugbymauri, 2015-02-25:
Yes i did already something like this in .NET. The approach will be the same.

DIY Version:
The plugin must implement an Interface Class, wich attaches the model and the gui (ex. main-window), of course the model and gui-elment should also implement a interface. Then you need a "Plugin-Loader" wich scans *,jar's in a directory and load them, then attche the model and the gui to the plugin.

Framework!
Otherway is to use an Rich-Client Framework, such as:
- RCP from Eclipse (learning curve),   Eclipse is build with RCP
- RCP Netbeans https://netbeans.org/features/platform/
These Frameworks provide already a Pluing-Arichtecture and much more but they are huge and the executable also. I also don't know if these frameworks works fluid on Boads such as RasperyPi

I'm not a JAVA expert, but i programmed some little apps in JAVA  (years ago) and did once an Ecplise RCP tutorial. But i think if you plan to reengineer USG i would go for an RCP-Frameworkm (Netbeans).

I think the GUI can not be a pluing. There is alway the main prgramm wich provides the interface to the GUI, then you can load the "UGS"-model, the main programm also provides the plugin mechnism. If you have the main programm the plugins can provide for ex. panels or menus which will be included in the main-GUI.
So for Android the only reusable part will be the UGS-Model.... if.....
You think the world really need to control a CNC machnine with an Android? me not!

winder, 2015-02-25:
This looks really promising: https://www.youtube.com/watch?v=BXQg_guFL2k

rugbymauri, 2015-02-26:
i quickly build a GUI Mockup with Netbean RCP

https://dl.dropboxusercontent.com/u/2710560/ugs2.zip

winder, 2015-03-03:
I've been reading up on Netbeans RCP, and experimenting. It's inspiring lots of ideas. Wrapping the UGS.jar file in a Netbeand library wrapper module was simple and work really well. The one annoyance is that updating the jar file needs to be done manually each time. Maybe the UGS.jar build script can be modified to copy the jar into the Netbeans Platform project.

There are a couple issues left that I need to investigate further:
1. How should the project be organized in git? I'd like to keep the "core" in a separate Netbeans project but still have everything in one git repository.

2. How will the build work? Somehow it needs to build UGS.jar, copy it into the Netbeans RCP project then build the Netbeans RCP project.There are probably lots of options here.

3. How will the final application be deployed? It doesn't look like a single executable jar is going to be possible anymore. At lease I'd like to have a single installer for all platforms which doesn't seem to be a default feature. There are a lot of options here, like "IzPack". Some of them seem to also install a compatible JRE which would be a nice feature.

winder, 2015-03-10:
I'm still thinking about (3), but I got the Netbeans Platform project committed to git and the build is now running on the CI server. The code is integrated to the point where you can connect to GRBL, send some jog commands, and monitor the console status. The visualizer is kind of in but there are some initialization glitches.

http://bit.ly/18xC2UQ

rugbymauri, 2015-03-10:
I looked to the UGS Platform. Nice!
(3) check "Project Properties" -> "Installer", then Project (right mouse click) -> Package as -> installer
:-)

rugbymauri, 2015-12-26:
i worked on an Non-Platform-UGS Plugin Architeture
see my fork of the UGS:  rugbymauri/Universal-G-Code-Sender
there is also a Sample Plugin here: rugbymauri/UGS-Plugin
just put the plugin JAR in the Plugin directory

basicly i implemented the following tutorial (sorry its in German) http://www.java-blog-buch.de/d-plugin-entwicklung-in-java/

winder, 2016-02-17:
@rugbymauri Really interesting that you were able to implement a plugin system. I'm pretty happy with the RCP approach now.

closed 2016-02-17 by winder