winder/Universal-G-Code-Sender #234: error on connection

dvdnhm, 2015-03-01:
I am getting the error below when I hit "Open" button and the app terminates.

java.lang.NullPointerException at com.willwinder.universalgcodesender.JSSCConnection.serialEvent(JSSCConnection.java:122) 
at jssc.SerialPort$EventThread.run(SerialPort.java:1112) com.willwinder.universalgcodesender.GUIBackend setControlState
INFO: Sending control state change.

closed 2015-03-01 by winder

dvdnhm, 2015-03-01:
this is all it shows, I ran it through command line the output ends with last info line

winder, 2015-03-01:
Thanks for the report, this should hopefully be fixed in the next nightly build.

dvdnhm, 2015-03-01:
I will test it once its available, thanks

dvdnhm, 2015-03-03:
it works now.

PS: I think I havent thanked for the great app you created yet, great job. thanks

commit a72ecf3f (2015-03-01): Check if there was data after reading the serial port. Closes #234