winder/Universal-G-Code-Sender #236: Visualizer not working; showing blank white screen on Windows 7 x64

NothinIsEasy, 2015-03-10:
The visualizer works for v1.0.6, but only displays a blank white window for versions v1.0.7 and v1.0.8.  

Using v1.0.8 (by running start-windows.bat), the command window is displaying these errors:

Stable Library
=========================================
Native lib Version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
Java lib Version   = RXTX-2.1-7
WARNING:  RXTX Version mismatch
        Jar version = RXTX-2.1-7
        native lib Version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
[JarClassLoader] INFO:  findResource() found: "com/jogamp/common/os/Platform.class" for caller null in codebase lib/glue
gen-rt.jar
[JarClassLoader] INFO:  findResource() found: "jogamp/common/Debug.class" for caller null in codebase lib/gluegen-rt.jar

[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/common/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/opengl/Debug.class" for caller null in codebase lib/jogl-all.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/opengl/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/newt/Debug.class" for caller null in codebase lib/jogl-all.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/newt/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "YF2000"
        at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
        at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
        at java.lang.Double.parseDouble(Unknown Source)
        at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.parseCoord(GcodePreprocessorUtils.java:340)
        at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.updatePointWithCommand(GcodePreprocessorUtil
s.java:192)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.handleGCode(GcodeParser.java:290)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.processCommand(GcodeParser.java:231)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:133)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:123)
        at com.willwinder.universalgcodesender.visualizer.GcodeViewParse.toObjRedux(GcodeViewParse.java:102)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.generateObject(VisualizerCanvas.java:443)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.setGcodeFile(VisualizerCanvas.java:175)
        at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.setGcodeFile(VisualizerWindow.java:93)
        at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1568)
        at com.willwinder.universalgcodesender.MainWindow.access$500(MainWindow.java:73)
        at com.willwinder.universalgcodesender.MainWindow$6.actionPerformed(MainWindow.java:310)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
[... 116 lines cut ...]
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NullPointerException
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.reshape(VisualizerCanvas.java:228)
        at jogamp.opengl.GLDrawableHelper.reshape(GLDrawableHelper.java:742)
        at jogamp.opengl.GLDrawableHelper.reshape(GLDrawableHelper.java:748)
        at javax.media.opengl.awt.GLCanvas$10.run(GLCanvas.java:1379)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
        ... 28 more

NothinIsEasy, 2015-03-10:
My machine is NOT an AMD64.  It's an Intel x64, so I don't know why it's looking for AMD64 libs.

winder, 2015-03-10:
Try the 2.0 classic GUI build, I just added a link to the main page. I made a few fixes to the visualizer recently and added some more log messages.

NothinIsEasy, 2015-03-10:
Thanks to the person who fixed the nightly build for 2.0, however, the visualizer for 2.0 isn't working either.  It displays a black window and prints the errors below.  I don't write Java code, but it seems like the .jar file maybe missing important platform specific resource files (at least that's what the comments from the [JarClassLoader] is stating).  Also, why are the program [JarClassLoader]comments about not finding "amd64" files?  I have an Intel x64 computer...  One last note:  These are the same errors I get in every version since v1.0.6.

INFO: Getting control state.
[JarClassLoader] INFO:  findResource() found: "com/jogamp/common/os/Platform.class" for caller null in codebase lib/glue
gen-rt.jar
[JarClassLoader] INFO:  findResource() found: "jogamp/common/Debug.class" for caller null in codebase lib/gluegen-rt.jar

[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/common/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/opengl/Debug.class" for caller null in codebase lib/jogl-all.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/opengl/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/newt/Debug.class" for caller null in codebase lib/jogl-all.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/newt/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
[JarClassLoader] INFO:  findResource() found: "jogamp/nativewindow/Debug.class" for caller null in codebase lib/jogl-all
.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/windows-amd64/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/nativewindow/windows/amd64/TAG.class"
Exception in thread "AWT-EventQueue-0" java.lang.NumberFormatException: For input string: "YF2000"
        at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
        at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
        at java.lang.Double.parseDouble(Unknown Source)
        at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.parseCoord(GcodePreprocessorUtils.java:340)
        at com.willwinder.universalgcodesender.gcode.GcodePreprocessorUtils.updatePointWithCommand(GcodePreprocessorUtil
s.java:192)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.handleGCode(GcodeParser.java:289)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.processCommand(GcodeParser.java:231)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:133)
        at com.willwinder.universalgcodesender.gcode.GcodeParser.addCommand(GcodeParser.java:123)
        at com.willwinder.universalgcodesender.visualizer.GcodeViewParse.toObjRedux(GcodeViewParse.java:102)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.generateObject(VisualizerCanvas.java:455)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.setGcodeFile(VisualizerCanvas.java:179)
        at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.setGcodeFile(VisualizerWindow.java:93)
        at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1683)
        at com.willwinder.universalgcodesender.MainWindow.access$1500(MainWindow.java:87)
        at com.willwinder.universalgcodesender.MainWindow$10.actionPerformed(MainWindow.java:498)
        at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
        at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
        at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
        at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
        at java.awt.Component.processMouseEvent(Unknown Source)
        at javax.swing.JComponent.processMouseEvent(Unknown Source)
        at java.awt.Component.processEvent(Unknown Source)
        at java.awt.Container.processEvent(Unknown Source)
        at java.awt.Component.dispatchEventImpl(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Component.dispatchEvent(Unknown Source)
        at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
        at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
        at java.awt.Container.dispatchEventImpl(Unknown Source)
        at java.awt.Window.dispatchEventImpl(Unknown Source)
[... cut ...]
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.init(VisualizerCanvas.java:201)
        at jogamp.opengl.GLDrawableHelper.init(GLDrawableHelper.java:640)
        at jogamp.opengl.GLDrawableHelper.init(GLDrawableHelper.java:662)
        at javax.media.opengl.awt.GLCanvas$9.run(GLCanvas.java:1366)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1275)
        ... 16 more
Mar 10, 2015 1:58:08 PM com.willwinder.universalgcodesender.visualizer.VisualizerCanvas reshape
INFO: Reshaping OpenGL context.

winder, 2015-03-10:
Having the screen be black is good, it means that the 3D libraries have been loaded. From your error it looks like you have a bad gcode, specifically "YF2000". I'm not sure if the standard allows it or not, but the UGS gcode parser can't handle it right now. Basically you are specifying that there is a Y coordinate, but then there is no coordinate before the feed rate is given.

Are you able to visualize another gcode file? There are several gcode files that I use to test here: https://github.com/winder/Universal-G-Code-Sender/tree/master/test_files

NothinIsEasy, 2015-03-10:
Ok.  I'll try the test files.  Here is the gcode file I was trying to visualize (this is the same file that  visualized just fine with v1.0.6).   It's a simple box. 

M136 (enable build)
M73 P0
G162 X Y F2000(home XY axes maximum)
G161 Z F900(home Z axis minimum)
G92 X0 Y0 Z-5 A0 B0 (set Z to -5)
G1 Z0.0 F900(move Z to '0')
G161 Z F100(home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
G92 X152 Y72 Z0 A0 B0
G1 X-112 Y-73 Z150 F3300.0 (move to waiting position)
G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
M109 S110 T0
M134 T0
M135 T0
M104 S230 T0
M133 T0
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
; 
; Makerbot Industries
; Miracle-Grue 2.4.1
; This file contains digital fabrication directives in gcode format
; For your 3D printer
; http://www.makerbot.com/support/makerware/documentation/slicer/
; 
; Active extruders in print: 0
; Chunk 0
; Position  0
; Thickness 0.2
; Width 0.4
G1 X105.400 Y-74.000 Z0.270 F9000.000 (Extruder Prime Dry Move)
G1 X-112 Y-73 Z0.270 F1800.000 E25.000 (Extruder Prime Start)
G92 A0 B0 (Reset after prime)
G1 Z0.000000 F1000
G1 X-112.0 Y-73.0 Z0.0 F1000 E0.0
G92 E0
G1 X-112.000 Y-73.000 Z0.000 F9000; Move to start position
G1 X-112.000 Y-73.000 Z0.000 F3600; Set speed for tool change
M135 T0; Extruder change
M104 T0 S230; Temperature Change
G1 X-112.000 Y-73.000 Z0.000 F1500 A-1.30000; Retract
G1 X-112.000 Y-73.000 Z0.200 F1380; Travel move
M73 P0; 
G1 X-11.900 Y-11.900 Z0.200 F9000; Travel move
G1 X-11.900 Y-11.900 Z0.200 F1500 A0.00000; Restart
G1 X-11.900 Y11.900 Z0.200 F1800 A0.83205; Inset
G1 X11.900 Y11.900 Z0.200 F1800 A1.66409; Inset
G1 X11.900 Y-11.900 Z0.200 F1800 A2.49614; Inset
G1 X-11.900 Y-11.900 Z0.200 F1800 A3.32818; Inset
G1 X-12.300 Y-12.300 Z0.200 F1800 A3.34796; Connection
G1 X-12.300 Y12.300 Z0.200 F1800 A4.20798; Outline
G1 X12.300 Y12.300 Z0.200 F1800 A5.06799; Outline
G1 X12.300 Y-12.300 Z0.200 F1800 A5.92800; Outline
G1 X-12.300 Y-12.300 Z0.200 F1800 A6.78802; Outline
M73 P1; 
G1 X-12.300 Y-12.300 Z0.200 F1500 A5.48802; Retract
G1 X-11.441 Y-11.614 Z0.200 F9000; Travel move
G1 X-11.441 Y-11.614 Z0.200 F1500 A6.78802; Restart
G1 X11.614 Y11.441 Z0.200 F1800 A7.92788; Infill
G1 X11.221 Y11.614 Z0.200 F1800 A7.94288; Infill
G1 X-11.614 Y-11.221 Z0.200 F1800 A9.07187; Infill
G1 X-11.614 Y-10.656 Z0.200 F1800 A9.09165; Infill
G1 X10.656 Y11.614 Z0.200 F1800 A10.19268; Infill
G1 X10.090 Y11.614 Z0.200 F1800 A10.21245; Infill
G1 X-11.614 Y-10.090 Z0.200 F1800 A11.28551; Infill
G1 X-11.614 Y-9.524 Z0.200 F1800 A11.30529; Infill
G1 X9.524 Y11.614 Z0.200 F1800 A12.35038; Infill
G1 X8.959 Y11.614 Z0.200 F1800 A12.37015; Infill
G1 X-11.614 Y-8.959 Z0.200 F1800 A13.38728; Infill
[... 1656 lines cut ...]
G1 X8.358 Y-11.614 Z25.000 F5400 A430.75085; Infill
G1 X11.614 Y-8.358 Z25.000 F5400 A430.91183; Infill
G1 X11.614 Y-8.924 Z25.000 F5400 A430.93161; Infill
G1 X8.924 Y-11.614 Z25.000 F5400 A431.06462; Infill
G1 X9.489 Y-11.614 Z25.000 F5400 A431.08439; Infill
G1 X11.614 Y-9.489 Z25.000 F5400 A431.18943; Infill
G1 X11.614 Y-10.055 Z25.000 F5400 A431.20921; Infill
G1 X10.055 Y-11.614 Z25.000 F5400 A431.28629; Infill
G1 X10.621 Y-11.614 Z25.000 F5400 A431.30606; Infill
G1 X11.614 Y-10.621 Z25.000 F5400 A431.35517; Infill
G1 X11.614 Y-11.186 Z25.000 F5400 A431.37494; Infill
G1 X11.186 Y-11.614 Z25.000 F5400 A431.39608; Infill
M73 P100; 
; End of print
G1 X11.186 Y-11.614 Z25.000 F1500 A430.09608; Short Retract Extruder: A
M18 A B(Turn off A and B Steppers)
G1 Z155 F900
G162 X Y F2000
M18 X Y Z(Turn off steppers after a build)
M109 S0 T0
M104 S0 T0
M73 P100 (end  build progress )
M70 P5 (We <3 Making Things!)
M72 P1  ( Play Ta-Da song )
M137 (build end notification)

NothinIsEasy, 2015-03-10:
I tested a couple of sample files from the link you gave me and both visualize using v2.0.  Thank you.  
The gcode I posted was generated from a MakerBot MakerWare program for a MakerBot Replicator 2x 3d printer... I guess I'm incorrectly assuming gcode is a standard?

winder, 2015-03-10:
UGS should be able to at least handle/ignore cases like this, I've just
never seen commands like "G162 X Y F2000" or "M18 X Y Z" where the X/Y/Z
weren't followed by a number.

NothinIsEasy, 2015-03-12:
I'm going to close this issue, thank you.

One last thing, if you have time.  Is there a list of G-Code commands that UGS is using?  Then I can, at least, pre-parse the gcode myself, to ferret out potential errors.

closed 2015-03-12 by NothinIsEasy

winder, 2015-03-12:
Its just the stand alone X Y Z, put a zero after them and I expect the unsupported commands will be ignored and everything else should look alright.

AbySet, 2019-01-10:
I'm using January 10 2019 nightly build version classic and visualizer is blank (white) and console is generating some Exception errors about OpenGL.
I'm not personly using classic version but i was doing some tests just to give some help.
My OS Windows10 x64 (All drivers and java are up to date).
I'll be glad to make any requested test if needed.
Here is the console content:

C:\Users\Phoenix\Desktop\UniversalGcodeSender-v2.0.0-stable2>java -jar -Xmx256m UniversalGcodeSender.jar
janv. 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFOS: Log location: C:\Users\Phoenix\.ugs\UniversalGcodeSender.json
janv. 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFOS: Loading settings.
Initializing...
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: JogPanel
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: JogPanel
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MacroPanel
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: CommandTextArea
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: ConfigureFirmwareAction$$Lambda$48/1413378318
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Initializing firmware... ...
/resources/firmware_config/
/resources/firmware_config/test_2.json
/resources/firmware_config/test_1.json
/resources/firmware_config/grbl.json
/resources/firmware_config/smoothieboard.json
/resources/firmware_config/tinyg.json
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: MainWindow
Jan 10, 2019 8:06:08 AM com.willwinder.universalgcodesender.model.GUIBackend addUGSEventListener
INFO: Adding UGSEvent listener: MainWindow
Jan 10, 2019 8:06:17 AM com.willwinder.universalgcodesender.model.GUIBackend connect
INFO: Connecting to GRBL on port COM4
Jan 10, 2019 8:06:17 AM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFO: Applying settings.
Jan 10, 2019 8:06:17 AM com.willwinder.universalgcodesender.AbstractCommunicator openCommPort
INFO: Connecting to controller using class: JSSCConnection with url jssc://COM4:115200
Jan 10, 2019 8:06:34 AM com.willwinder.universalgcodesender.model.GUIBackend sendGcodeCommand
INFO: Sending gcode command: $$
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.model.GUIBackend setGcodeFile
INFO: Setting gcode file.
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Start preprocessing
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.model.GUIBackend preprocessAndExportToFile
INFO: Preprocessing C:\Users\Phoenix\Desktop\CNC_Stuff\CNC_Files\For_Construction\spindle_plate.cnc to C:\Users\Phoenix\AppData\Local\Temp\1547104006158-0\spindle_plate.cnc_ugs_1547104006158
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFO: Took 261ms to preprocess
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.model.GUIBackend getProcessedGcodeFile
INFO: Getting processed gcode file.
Jan 10, 2019 8:06:46 AM com.willwinder.universalgcodesender.utils.SettingsFactory saveSettings
INFO: Saving settings.
Object bounds: X (0.0, 115.4444)
               Y (0.0, 116.2448)
               Z (0.0, 19.73)
Center = (57.7222, 58.1224, 9.865)
Num Line Segments :2194
Jan 10, 2019 8:06:54 AM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFO: Adding controller listener: VisualizerWindow
Jan 10, 2019 8:06:54 AM com.willwinder.universalgcodesender.visualizer.VisualizerCanvas init
INFO: Initializing OpenGL context.
Object bounds: X (0.0, 115.4444)
               Y (0.0, 116.2448)
               Z (0.0, 19.73)
Center = (57.7222, 58.1224, 9.865)
[... 101 lines cut ...]
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.awt.EventQueue$4.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
[... cut ...]
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.renderModel(VisualizerCanvas.java:388)
        at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.display(VisualizerCanvas.java:297)
        at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
        at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
        at com.jogamp.opengl.awt.GLCanvas$11.run(GLCanvas.java:1424)
        at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
        ... 28 more