winder/Universal-G-Code-Sender #238: Universal Gcode Sender Startup Units of Measure

pjakab, 2015-03-17:
When UGS comes up, the "Machine Control" tab shows inches as the default units of measure. The inches radio button is selected. This happens regardless of whether you have a GRBL setup to do a G21 when it initializes via a previous $N0=G21. And UGS will send, for example, G20 G91 G0 Y1, regardless of the current GRBL units of measure state. Obviously UGS is not aware, nor does it care, of the current GRBL units of measure state, and I think that part is OK, UGS's radio button has inches selected, so it sends an inches command.

What I'm asking for is a way to tell UGS to use mm as a default instead of inches, to let the user pick what units of measure they prefer to work with and remember them to use as a default in subsequent invocations.

I believe this could easily be implemented (if not already there, and I don't know about it) with a preferences file, or command line argument.

Thank you.

pjakab, 2015-03-18:
I was looking at the source code and it seems mm is already the default units of measure.

Deleting the json file, and then starting UGS does indeed come up with the mm radio button selected. However the next invocation it's back to inches, and the json file has "defaultUnits": "inch".

I never see the UniversalGcodeSender.properties file. I'm very tempted to bring the source into NetBeans and see if I can figure it out. I'm still looking at the source files to see why the defaultUnits property is being changed to inches.

pjakab, 2015-03-18:
Sorry, closed by mistake.

winder, 2015-03-18:
The json file should get updated while UGS is shutting down,but its possible that I forgot to save this value when adding the feature.

pjakab, 2015-03-18:
Hi Will, thanks for the response, I just tried again and this is exactly what happens:

1) With no json file, UGS opens the mm radio button is selected.
2) Close UGS, the json file gets created with "defaultUnits": "mm". So far so good.
3) Open UGS again, inches radio button is selected, json file still says "defaultUnits": "mm"
4) Close UGS, json files gets updated, but now "defaultUnits": "inch"

From then on, until the json file is deleted, "defaultUnits": "inch" is not changed anymore.

So looks like things go wrong on step 3 when UGS comes up, the json file says "defaultUnits": "mm", but the inches radio button is selected anyhow.

Hope this helps.

...Peter

pjakab, 2015-03-18:
I tried this scenario with 2.0 and it works as expected, the default comes up in mm, and whatever is selected by the user is properly preserved across invocations of the program.

Knowing that it works in 2.0, should make it easy to fix on 1.8, right ;-)

Please let me know if you do plan to fix it in 1.8. Or alternatively if you consider 2.0 good enough to use.

Thanks for listening.

winder, 2015-03-18:
It works in the 2.0 classic gui or the platform? Classic gui should be fine
to use, the platform is still missing functionality.

pjakab, 2015-03-18:
Works in the Classic GUI. The radio buttons on the platform aren't working well yet,both can be selected at the same time. 

Thanks for your help. I'll start using 2.0 then.

closed 2015-03-19 by winder