winder/Universal-G-Code-Sender #240: Sends CNC cutting off in wrong direction Loss of 2 $50 sheets of material.

anthonyzeh, 2015-03-20:
I start to send the job after resetting zero, It moves into place then just takes a diagonal cut across my material! Not following pattern in visualizer when used. I thought maybe the visualizer was going wrong and bugging out the system. Sometimes it works and sometimes it does this weird random direction cut across my material on startup. Going to lose a large job because of this and I will not be out of pocket to replace their materials.

winder, 2015-03-20:
What do you do after clicking the "reset zero" button? What version of GRBL are you using?

UGS just sends commands to GRBL when you click these buttons, there are different commands supported on GRBL v8 and v9:
    GCODE_RESET_COORDINATES_TO_ZERO_V9 = "G10 P0 L20 X0 Y0 Z0";
    GCODE_RESET_COORDINATES_TO_ZERO_V8 = "G92 X0 Y0 Z0";

    GCODE_RESET_COORDINATE_TO_ZERO_V9 = "G10 P0 L20 %c0";
    GCODE_RESET_COORDINATE_TO_ZERO_V8 = "G92 %c0";

No need to lose a job because this free software isn't doing what you need it to do, if you need to run a different command, just send the command directly instead of using the UGS buttons:
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

anthonyzeh, 2015-03-20:
I am running GRBL 8, the reset command is ok, Just that for some reason my machine is taking off in a random diagonal. Doesn't seen to correlate to anything else that I can see. I tried GBRL 9 but when the unit returns and I tell it to go back to zero the z axis goes up instead of down. So I changed back to 8 because it was more predictable (until now). I would like to get 9 working when I have time to figure out the issues in that one.

winder, 2015-03-20:
Return to zero raises the tool head before going back to zero in case the job finishes while buried in the material. I think that should happen for v0.8 and v0.9 though.

What are you doing when the machine takes off diagonally?
Could you explain all the operations you do after connecting to GRBL?

anthonyzeh, 2015-03-20:
After setting my unit to zero manually to my material. I turn on the power to the GRBL and open the connection. I reset to zero, load the gcode, then hit send. It starts to cut a pattern then, takes off, and sometimes doing the same procedure, it does great. I have been manually setting it because sometimes when I tell it to go home it will work just fine and sometimes it will take off or plunge into my waste board. Been dealing with a bit of unpredictability.

As far as the 8 and 9 part with the head going up, that I see, but in my version 9 it only went up and didn't go back down to the zero location.

winder, 2015-03-20:
You might try posting on the ShapeOko forum, I seem to recall similar issues. A quick search for "diagonal" found [this](http://www.shapeoko.com/forum/viewtopic.php?f=6&t=5698&p=42674&hilit=diagonal#p42674) thread which looks like a possible match.
Are there any G40 commands in your gcode file?

anthonyzeh, 2015-03-20:
I checked for G40 commands, there were none. Maybe there is feed back from the material causing a spike which is throwing it off. I actually had issues when I was running dust collection with an anti-static hose. The material I am working with is acrylic mirror. could be a static impulse throwing a loop into the problem. Thanks for your feed back. I will just take a loss on the job before I get more in debt. 

Any ideas on the grbl 9 why it goes up when going to zero, but not actually going back down to the zero point for the Z axis? Thanks!

winder, 2015-03-20:
Are you using v0.8 or v0.8c?

anthonyzeh, 2015-03-20:
V0.8c.

winder, 2015-03-20:
I'm not sure then, there is a difference in UGS for v0.8 and v0.8c, but v0.8c and v0.9+ should use the same commands:

G21 G90 G0 Z4.0
G90 G28 X0 Y0

anthonyzeh, 2015-03-20:
Thanks, I guess I will have to stick with 8c since it's been a little more predictable. at least until this job. 

I appreciate your time.
Tony

JanJansen47, 2015-03-21:
Jan Jansen
Sagittalaan 22
5632 AL Eindhoven
+31(0)6  888  06529 

  
 

Date: Fri, 20 Mar 2015 12:30:23 -0700

nsiatras, 2015-04-12:
I had the same problem with GCode generated with MeshCam.
After a few tests I find out that the problem happens when I choose the "Use Parallel Path" option in meshcam toolpath parameters.

If I don't then the GCode runs fine :)

closed 2015-10-14 by winder

breiler, 2023-12-08:
@Sheshu1987 this was resolved in 2015. If you have a new problem I suggest that you create a new issue and provide enough details for us to reproduce it.

What CAM software are you using?
A sample program?

I don't know of any GRBL version 1.7.0.2, if it is a forked version we need to know what brand of controller you are using.