winder/Universal-G-Code-Sender #243: Resume work after a power outage

trNewDanger, 2015-03-29:
The case of power outage would be the most extremist problem, but it would be convenient to have a way to pause and save work in process to resume later, perhaps a auto save every few minutes or something like that would be useful.
There is a viable solution for this?

EvansMike, 2015-04-01:
I was wondering myself what happens on a machine emergency stop, the PC being unaffected.  I assume (hah!) that since OK reponses are expected after each gcode is sent (or is it after it's been executed?) the program should be able to resume on (or back one) the last line executed? After manual intervention of course.

philreindl, 2016-03-05:
I imagine it wouldn't be terribly difficult to add a "start from line X" feature.  The challenge would be to ensure that machine position is preserved and parser state is reserved.  If (for example) units were to have changed part way through the file, it would be critical for the unit change step to not be omitted in the resume.  I know there are other cases where it would be bad to blindly skip over some commands when resuming.  But I'm not confident I'd be able to identify them.

winder, 2016-03-05:
@philreindl I hadn't considered the gcode state, good point that does complicate things. Depending on how the gcode is written even the "G" commands can be omitted (G1 X1, followed by X2Y2 is a valid way to send two commands). The gcode parser implements most of this though, so it would be a matter of parsing up to the requested line, moving to the starting point in a safe way, sending out all the state information before resuming then resuming at the correct (possibly modified to include a G-command) line.

Also it looks like this is a duplicate of #123, I missed it while cleaning things up recently.

closed 2016-03-05 by winder