winder/Universal-G-Code-Sender #244: Small pauses after 20.000 lines of Gcode

raulgotor, 2015-03-31:
Hi!

Thanks for this great job! I've love it!

I have problems with bigger files on a Mac, Snow Leopard, UGS 1.0.6. I have increased virtual machine memory up to 4GB, but still, after some 18.000 - 20.000 lines machine randomly pauses (for 1 second approximately). This pauses becomes more and more frequent as the Gcode program continues.

Any clue?

Thanks in advance!

Raúl

rugbymauri, 2015-03-31:
Hi Raul

i had same problems with BIG G-code files. with an 115000 lines files the UGS freezes.
Loading of the code takes long time.

@winder 
i supposed the "Estimated Time Remaing" counter is the problem. so i removed this part of the code (on my fork). since then i never tried with so big files.
ETR is a nice feature. But really needed?

winder, 2015-03-31:
This is a known issue. UGS loads the entire file into memory during processing and can exhaust the memory available to it.

Two things you can do to help are:
1) Leave the Visualizer closed.
2) Run the UGS jar file with -Xmx, see [this doc](http://docs.oracle.com/cd/E13222_01/wls/docs81/perform/JVMTuning.html#1115453).

A more permanent fix is planned with issue #168 

@rugbymauri The "Estimated Time Remaining" adds very little overhead, all it does is calculate the average time spent on each line of gcode and use that to create an estimate based on the remaining number of lines.

winder, 2015-03-31:
Also I have added a number of memory related improvements since v1.0.6, so you should upgrade to v1.0.7 or v1.0.8.

closed 2015-03-31 by winder

raulgotor, 2015-03-31:
Hi Will,
Yes, I've seen #168. I'm running the program with -Xmx4096m

Unluckily, v1.0.7 and v1.0.8 requieres Java 7, which is not supported on Snow Leopard.

Thanks anyway!

philreindl, 2015-10-07:
I've had some success by adding the command line option 
-XX:+UseConcMarkSweepGC