winder/Universal-G-Code-Sender #245: error opening connection (gnu.io.PortInUseException)

victornpb, 2015-04-07:
I'm getting random error opening connection (gnu.io.PortInUseException).
I already issued the 2 lines to create a lock folder and chmod it.

After closing the application, or after a crash it does not recover, when I open it again. Already tried killing the JarLauncher and Boot process.

I have to restart my computer and it start working again.

OS X 10.10.2 Java 7 60

winder, 2015-04-07:
Which version of UGS are you using?

victornpb, 2015-04-07:
1.0.7

winder, 2015-04-07:
To recover without rebooting you need to delete the contents of /var/lock

If the problems continue, you could also try the 2.0 classic gui release which is using a new serial library, it might have better support for your hardware.

closed 2016-01-12 by winder