winder/Universal-G-Code-Sender #249: GCode visualizer window doesn't work

roberto-arista, 2015-04-14:
Hi!

I upgraded the grbl firmware from 0.8c to 0.9g.
When I started to run a job with UGCS 1.0.6 the console output me some errors, so I upgraded to UGCS 1.0.8 too. Now my shapeoko2 runs fine, but I am not able to visualize gCode before sending it.
[image]
(If I click visualize nothing happens)

What kind of informations shall I provide you with to fix this issue?
Is there maybe something I should take care? (I updated to the latest java version)

Thanks

winder, 2015-04-14:
If you launch UGS from a command line there should be some extra information.

Also if you could try the nightly build: http://bit.ly/1vIcB9E

roberto-arista, 2015-04-20:
Hi!
Sorry for the delay, I've been far from my machine.
I tried a nightly build before posting and the issue was identical.
Here it is the output from the console:
```
MacBook-Pro-di-Roberto-3:UniversalGcodeSender robertoarista$ ./start.sh
[Boot] INFO:  myJarPath=file:/Users/robertoarista/Desktop/UniversalGcodeSender/UniversalGcodeSender.jar
[JarClassLoader] INFO:  setProperties(com.simontuffs.onejar.JarClassLoader@60e53b93)
[JarClassLoader] INFO:  oneJarPath=file:/Users/robertoarista/Desktop/UniversalGcodeSender/UniversalGcodeSender.jar
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:  oneJarPath=file:/Users/robertoarista/Desktop/UniversalGcodeSender/UniversalGcodeSender.jar
[JarClassLoader] INFO:  findResource(): unable to locate "resources/MessagesBundle_en.properties"
apr 20, 2015 11:20:53 AM com.willwinder.universalgcodesender.SettingsFactory loadSettings
INFORMAZIONI: Loading settings.
apr 20, 2015 11:20:53 AM com.willwinder.universalgcodesender.SettingsFactory loadSettings
INFORMAZIONI: Settings file location: /Users/robertoarista/Library/Preferences/UniversalGcodeSender.json
Experimental:  JNI_OnLoad called.
Stable Library
=========================================
Native lib Version = RXTX-2.1-7
Java lib Version   = RXTX-2.1-7

[JarClassLoader] INFO:  findResource() found: "com/jogamp/common/os/Platform.class" for caller null in codebase lib/gluegen-rt.jar
[JarClassLoader] INFO:  findResource() found: "jogamp/common/Debug.class" for caller null in codebase lib/gluegen-rt.jar
[JarClassLoader] INFO:  findResource(): unable to locate "natives/macosx-universal/"
[JarClassLoader] INFO:  findResource(): unable to locate "jogamp/nativetag/common/macosx/universal/TAG.class"
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: /System/Library/Frameworks/gluegen-rt.Framework/gluegen-rt
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1825)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibraryInternal(JNILibLoaderBase.java:596)
	at com.jogamp.common.jvm.JNILibLoaderBase.access$000(JNILibLoaderBase.java:64)
	at com.jogamp.common.jvm.JNILibLoaderBase$DefaultAction.loadLibrary(JNILibLoaderBase.java:96)
	at com.jogamp.common.jvm.JNILibLoaderBase.loadLibrary(JNILibLoaderBase.java:459)
	at com.jogamp.common.os.DynamicLibraryBundle$GlueJNILibLoader.loadLibrary(DynamicLibraryBundle.java:388)
	at com.jogamp.common.os.Platform$1.run(Platform.java:209)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.common.os.Platform.<clinit>(Platform.java:179)
	at javax.media.opengl.GLProfile.<clinit>(GLProfile.java:120)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:263)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:230)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:217)
	at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.<init>(VisualizerCanvas.java:131)
	at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.<init>(VisualizerWindow.java:71)
	at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1543)
	at com.willwinder.universalgcodesender.MainWindow.access$500(MainWindow.java:73)
	at com.willwinder.universalgcodesender.MainWindow$6.actionPerformed(MainWindow.java:310)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
[... 79 lines cut ...]
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

winduptoy, 2015-04-21:
Same problem here on Mac OS 10.10.3.

winder, 2015-04-21:
I found a couple files that might have been missing from the mac version. Does this build work? https://winder.ci.cloudbees.com/job/UGS/264/artifact/release/UniversalGcodeSender.zip

RayMJr, 2015-04-21:
I have the same visualization issue that Roberto had.
The build at the posted link above seems to have fixed the issue!

Thank you for the continued good work & progress!

Ray

roberto-arista, 2015-04-22:
It works!
–––
Roberto Arista
http://robertoarista.it <http://robertoarista.it/>
–––
Mobile BE: +32 494 71 38 56
Mobile IT: +39 366 4537413
Skype: jimbo475
–––
My public PGP key is here:
http://projects.robertoarista.it/robertoArista_publicKey.txt <http://projects.robertoarista.it/robertoArista_publicKey.txt>
–––

winder, 2015-04-22:
Thanks everyone.

closed 2015-04-22 by winder

klajo, 2015-07-13:
Hi,

I get the same "Can't load library: /System/Library/Frameworks/gluegen-rt.Framework/gluegen-rt" problem as roberto-arista above on 1.0.8 on OSX.  1.0.7 works fine.  The https://winder.ci.cloudbees.com/job/UGS/264/artifact/release/UniversalGcodeSender.zip link is dead.

I downloaded http://bit.ly/1BSKon6 (from README.md rev 966b93a).  This contains:

$ jar -tvf UniversalGcodeSender.jar | grep "gluegen"
287502 Wed Feb 25 06:34:08 CET 2015 lib/gluegen-rt.jar
700421 Wed Feb 25 06:34:08 CET 2015 lib/gluegen.jar
  9092 Wed Feb 25 06:34:08 CET 2015 binlib/arm/libgluegen-rt.so
  8256 Wed Feb 25 06:34:08 CET 2015 binlib/linux32/libgluegen-rt.so
 10016 Wed Feb 25 06:34:08 CET 2015 binlib/linux64/libgluegen-rt.so
 35296 Wed Feb 25 06:34:08 CET 2015 binlib/macosx/libgluegen-rt.jnilib
 16896 Wed Feb 25 06:34:08 CET 2015 binlib/windows32/gluegen-rt.dll
 18944 Wed Feb 25 06:34:08 CET 2015 binlib/windows64/gluegen-rt.dll

Do I have to get it from someplace else or do something to run this?

Thanks,
Klas

winder, 2015-07-13:
You'll have to download the nightly build to get the fix.

Your grep shows the problem:
```shell
$ jar -tvf UniversalGcodeSender.jar | grep "gluegen"287502 Tue Jul 07 18:26:50 EDT 2015 lib/gluegen-rt.jar
700421 Tue Jul 07 18:26:50 EDT 2015 lib/gluegen.jar
  9092 Tue Jul 07 18:26:50 EDT 2015 binlib/arm/libgluegen-rt.so
  8256 Tue Jul 07 18:26:50 EDT 2015 binlib/linux32/libgluegen-rt.so
 10016 Tue Jul 07 18:26:50 EDT 2015 binlib/linux64/libgluegen-rt.so
 35296 Tue Jul 07 18:26:50 EDT 2015 binlib/macosx/libgluegen-rt.dylib <-- this was missing
 35296 Tue Jul 07 18:26:50 EDT 2015 binlib/macosx/libgluegen-rt.jnilib
 16896 Tue Jul 07 18:26:50 EDT 2015 binlib/windows32/gluegen-rt.dll
 18944 Tue Jul 07 18:26:50 EDT 2015 binlib/windows64/gluegen-rt.dll
```

klajo, 2015-07-13:
Hi,

Yes, thanks, I noticed the missing file.  Can perhaps the problem be
mentioned for 1.0.8 on OSX, or fixed?

Cheers,
Klas

winder, 2015-07-13:
Sorry, but I don't backport fixes. The nightly build contains the fix. You could also patch the jar file if you don't want to upgrade.

klajo, 2015-07-13:
Ok, thanks.

papillion47, 2024-10-01:
Merhaba, uzun zamandır yeni ve hızlı çalışan PC' mde yüklü bulunan VectricAspire ile hazırladığım gravürlerimi yine aynı yeni PC' mde GRBL ile işlemekteydim. Ne olduysa şu anda GRBL nin "Visualizer" denilen kısmında yapılacak iş ve dolayısıyla simülasyon kayboldu gözükmüyor. Diğer eski bilgisayarımda hiç problem yok, GRBL de 3018 cihaz da kusursuz çalışıyor. Defalarca sorunun GRBL programında olduğunu düşünerek yeniden yükledim, sonuç aynı, Visualizer çalışmıyor.. Buna ne sebep oldu? Bir yardımınız veya bu konuda yardım kaynağı verebilirseniz çok mutlu olacağım. Saygı ve selâmlarla..

breiler, 2024-10-01:
@papillion47 please create a new issue.

papillion47, 2024-10-29:
GRBL visualizer halâ çalışmıyor. Bu arada defalarca GRBL yi kaldırıp yeniden yükledim. CNC kartını sipariş edip yeniden yükledim. Konsol dan yararlanarak olması gereken kodları yeniden yükledim. Jog komutları, spindle, diğer fonksiyonlar doğru çalışıyor. GRBL Hiçbir hata mesajı vermiyor.. Sadece "visualizer" görev yapmıyor. Rastgele bir tap uzantılı basit bir dosya açıyorum. Visualizer de görünmüyor ama CNC de işleme başlatılıyor. Normal çalışıyor. Bunalım içindeyim.. Bu derdime bir çare gösterebilecek olan yok mu?..

breiler, 2024-10-29:
@papillion47 I might be able to help if you create a new issue with all details. Cross posting to old closed unrelated issues will make things very difficult for us.