winder/Universal-G-Code-Sender #254: NullPointerException generated by Localization.getString()

soundstorm, 2015-05-07:
```
java.lang.NullPointerException
	at com.willwinder.universalgcodesender.MainWindow.setStatusColorForState(MainWindow.java:2153)
	at com.willwinder.universalgcodesender.MainWindow.statusStringListener(MainWindow.java:2512)
	at com.willwinder.universalgcodesender.AbstractController.dispatchStatusString(AbstractController.java:730)
	at com.willwinder.universalgcodesender.GrblController.handlePositionString(GrblController.java:421)
	at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:145)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:717)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListener(AbstractCommunicator.java:203)
	at com.willwinder.universalgcodesender.AbstractCommunicator.dispatchListenerEvents(AbstractCommunicator.java:177)
	at com.willwinder.universalgcodesender.GrblCommunicator.responseMessage(GrblCommunicator.java:193)
	at com.willwinder.universalgcodesender.connection.JSSCConnection.serialEvent(JSSCConnection.java:132)
	at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1299)
```

GrblUtils.java may (and does for me) return null. Maybe insert a check in GrblController.java on `handlePosition(String)` if the value fetched is == null?

winder, 2015-05-10:
What localization are you using? I made a change that might fix the problem, does this build work for you? https://winder.ci.cloudbees.com/job/UGS/lastSuccessfulBuild/artifact/release/UniversalGcodeSender.zip

soundstorm, 2015-05-10:
Seems to be working right now.
I’m using OS X – switched from EN to DE and back, as the formatting in german is not that well. May investigate some time on translation, if I’m done with my project. Thank you!

soundstorm, 2015-05-10:
Okay, was to quick... Just got a longer list of NullPointers:
```
May 10, 2015 5:32:39 PM com.willwinder.universalgcodesender.model.GUIBackend getFile
INFORMATION: Getting gcode file.
May 10, 2015 5:32:39 PM com.willwinder.universalgcodesender.model.GUIBackend getControlState
INFORMATION: Getting control state.
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.getSendDuration(GUIBackend.java:319)
	at com.willwinder.universalgcodesender.MainWindow$53$1.run(MainWindow.java:1730)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.getSendDuration(GUIBackend.java:319)
	at com.willwinder.universalgcodesender.MainWindow$53$1.run(MainWindow.java:1730)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.getSendDuration(GUIBackend.java:319)
	at com.willwinder.universalgcodesender.MainWindow$53$1.run(MainWindow.java:1730)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.getSendDuration(GUIBackend.java:319)
	at com.willwinder.universalgcodesender.MainWindow$53$1.run(MainWindow.java:1730)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:744)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
[... 1884 lines cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
[... cut ...]
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:714)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

soundstorm, 2015-05-10:
Right after canceling a file transmission:
```
java.lang.NullPointerException
	at com.willwinder.universalgcodesender.AbstractController.commandSent(AbstractController.java:627)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListener(AbstractCommunicator.java:194)
	at com.willwinder.universalgcodesender.AbstractCommunicator.dispatchListenerEvents(AbstractCommunicator.java:177)
	at com.willwinder.universalgcodesender.GrblCommunicator.streamCommands(GrblCommunicator.java:152)
	at com.willwinder.universalgcodesender.GrblCommunicator.responseMessage(GrblCommunicator.java:203)
	at com.willwinder.universalgcodesender.connection.JSSCConnection.serialEvent(JSSCConnection.java:132)
	at jssc.SerialPort$LinuxEventThread.run(SerialPort.java:1299)
```
I'm using a Leonardo to parse Information and displaying them on a LCD. May the USB-Serial communication of the 32u4 Chip be the problem? But as the 16u2 uses the same USB-Interface, the UNO users would have the same error.

winder, 2015-09-14:
Sorry for the delay. Is this problem still happening for you? I've made a lot of changes since May which might have helped.

closed 2016-01-12 by winder

rajsenthil, 2016-12-31:
I am running the latest version v1.0.9 with jdk 8. I am getting the repeated NullPointerException
`
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.willwinder.universalgcodesender.MainWindow.setStatusColorForState(MainWindow.java:2201)
        at com.willwinder.universalgcodesender.MainWindow.statusStringListener(MainWindow.java:2552)
        at com.willwinder.universalgcodesender.AbstractController.dispatchStatusString(AbstractController.java:757)
        at com.willwinder.universalgcodesender.GrblController.handlePositionString(GrblController.java:414)
        at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:126)
        at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:743)
        at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListener(AbstractCommunicator.java:198)
        at com.willwinder.universalgcodesender.AbstractCommunicator$1.run(AbstractCommunicator.java:165)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)
`

rajsenthil, 2016-12-31:
Few more lines of exception

`Dec 30, 2016 6:57:52 PM com.willwinder.universalgcodesender.model.GUIBackend connect
INFO: Connecting to GRBL on port COM3
Dec 30, 2016 6:57:52 PM com.willwinder.universalgcodesender.model.GUIBackend sendControlStateEvent
INFO: Sending control state change.
Dec 30, 2016 6:57:52 PM com.willwinder.universalgcodesender.model.GUIBackend getFile
INFO: Getting gcode file.
Dec 30, 2016 6:57:52 PM com.willwinder.universalgcodesender.model.GUIBackend getControlState
INFO: Getting control state.
Dec 30, 2016 6:57:52 PM com.willwinder.universalgcodesender.model.GUIBackend getFile
INFO: Getting gcode file.
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
        at com.willwinder.universalgcodesender.MainWindow.setStatusColorForState(MainWindow.java:2201)
        at com.willwinder.universalgcodesender.MainWindow.statusStringListener(MainWindow.java:2552)
        at com.willwinder.universalgcodesender.AbstractController.dispatchStatusString(AbstractController.java:757)
        at com.willwinder.universalgcodesender.GrblController.handlePositionString(GrblController.java:414)
        at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:126)
        at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:743)
        at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListener(AbstractCommunicator.java:198)
        at com.willwinder.universalgcodesender.AbstractCommunicator$1.run(AbstractCommunicator.java:165)
        at java.awt.event.InvocationEvent.dispatch(Unknown Source)
        at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
        at java.awt.EventQueue.access$500(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.awt.EventQueue$3.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
        at java.awt.EventQueue.dispatchEvent(Unknown Source)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
        at java.awt.EventDispatchThread.run(Unknown Source)`