winder/Universal-G-Code-Sender #255: UGS 1.0.8/2.0 Won't open in Windows 7

SWWC, 2015-05-08:
Newbie Shapeoko 2 owner here. Recently updated Grbl to v0.9i. UGS v1.0.7 opens but doesn't support any zeroing commands. I proceeded to download UGS v1.0.8, but trying to execute the .jar leaves me with a blank screen. After Downloading 2.0 the same thing occurs. Any suggestions, fixes, or hints in the right direction? I was chosen to set up a Shapeoko 2 to test for my high school manufacturing class. I don't have any experience in coding as of yet so please be nice and keep it KISS. c:

*I read a similar issue and it was mentioned I should update grblutils..any idea how to do that?

winder, 2015-05-08:
If you launch UGS from a command prompt there should be some error messages. Could you post them here?

SWWC, 2015-05-13:
C:\Users\CNC2\Downloads\UniversalGcodeSender-v1.0.8>java -jar -Xmx256m Universal
GcodeSender.jar
[Boot] INFO:  myJarPath=file:/C:/Users/CNC2/Downloads/UniversalGcodeSender-v1.0.
8/UniversalGcodeSender.jar
[JarClassLoader] INFO:  setProperties(com.simontuffs.onejar.JarClassLoader@be858
)
[JarClassLoader] INFO:  oneJarPath=file:/C:/Users/CNC2/Downloads/UniversalGcodeS
ender-v1.0.8/UniversalGcodeSender.jar
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:  oneJarPath=file:/C:/Users/CNC2/Downloads/UniversalGcodeS
ender-v1.0.8/UniversalGcodeSender.jar
[JarClassLoader] INFO:  findResource(): unable to locate "resources/MessagesBund
le_en.properties"
May 13, 2015 9:03:23 AM com.willwinder.universalgcodesender.SettingsFactory load
Settings
INFO: Loading settings.
May 13, 2015 9:03:23 AM com.willwinder.universalgcodesender.SettingsFactory load
Settings
INFO: Settings file location: C:\Users\CNC2\UniversalGcodeSender.json
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.simontuffs.onejar.Boot.run(Boot.java:315)
        at com.simontuffs.onejar.Boot.main(Boot.java:152)
Caused by: java.lang.NullPointerException
        at com.willwinder.universalgcodesender.MainWindow.initProgram(MainWindow
.java:2059)
        at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java
:84)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:1
963)
        ... 6 more

Hope this helps

Jimmydaux, 2015-05-13:
I had this issue on my windows 8.1 machine. 1.0.8 was working fine, but then stopped working one day and 2.0 wouldn't work at all. I don't know if this is the correct solution and don't know if there are further implications to this method, but by deleting the UniversalGcodesender.json file and universalgcodesender.properties file at C:\Users\YourComputerName, it corrected the problem for me and 2.0 now launches. I noticed that the .json file was 0 bytes before and since deleting it and allowing the program to recreate, it is now 2 kb. I believe the file somehow got corrupted and a deletion and subsequent recreation resolved my issue.

SWWC, 2015-05-18:
Thanks for the tip! My .json file was 0bytes as well. Deleting them seemed to fix the issue. Currently running a 2.0 build

closed 2015-05-18 by SWWC