winder/Universal-G-Code-Sender #258: "Return to zero" has weird behavior

RayMJr, 2015-06-05:
When I execute the "Return to zero" command from the UGS machine control interface, it changes the machine units to metric, moves the z axis to the work coordinate zero, moves the x & y axis to work coordinates zero, then moves all three axises to machine position zero.

On some setups all  of that could cause some damage.
Is there a way to tame that down to go either to machine zero, or work coordinates zero?

And I don't think it should change the machine units!

Thank you 

Ray

winder, 2015-06-05:
The intended behavior is that the tool is raised to some Z height that has been seen before and then moved to zero. I didn't realize there was the weird behavior you're describing when using work coordinates.

The unit change was supposed to be a safety feature for the tool raise part, in case the machine was at 10mm and then the units changed to inches.

UGS uses these commands:
G21 G90 G0 Z<value>
G90 G28 X0 Y0

What do you think should be used?

Jibeji, 2015-06-07:
Hi,
Same behavior here.
Shoudn't it go to the Work zero position instead of Machine zero position ?
I think so.

Here is an example.

First, $H to go to Home position:
[image]
Both Work and Machine positions are set to zero.

Second, defining the zero for Work position:
[image]
 Work position is set to zero, whatever the machine position. Correct.

Third, click on Return to zero:
[image]
It goes first to Work position, and after a few miilisec goes back to Machine zero position.

RayMJr, 2015-06-09:
After reading up on the G28 code, I understand what it is up to.
I have not ever used G28 or set G28.1, the the behavior I am seeing is what
should happen when the "Return to zero"  commands from UGS are sent.

I would never change units while in a session!

I would expect a return to zero to go to zero of the current work
coordinates (G54 default), or to the machine coordinates (G53). NOT BOTH!
The move up in Z before any X or Y movement is good.

So If I were to choose, here it is:

G53 G0 Z0
G53 G0 X0 Y0
m5
m30

The way I run my machine, that moves he quill up, then  moves the spindle
away from the user.

Thank you!!

Ray

--J. Ray Mitchell Jr.
jrmitchellj@gmail.com
(818)324-7573

The dogmas of the quiet past, are inadequate to the stormy present. The
occasion is piled high with difficulty, and we must rise with the occasion.
As our case is new, so we must think anew and act anew. We must disenthrall
ourselves, and then we shall save our country.*Abraham Lincoln
<http://www.quotationspage.com/quotes/Abraham_Lincoln/>*, *Annual message
to Congress, December 1, 1862*
*16th president of US (1809 - 1865)*

tapnair, 2015-06-09:
I agree with Ray G28 is tough for some users that don't know they have to have a set position with G28.1

I actually just went through this myself and then realized I had never set a logical G28 Position.  

Having an easy way to return to work/machine zero seems like a logical thing to me?

As a new user seeing a button that says "return to zero" then does not go to either of the places to which I see coordinates in the UI (Machine and Work) made me think there was something wrong.  I am sure that for experienced users this may make sense but it is not apparent to the new user in the interface.

closed 2015-06-14 by winder

RayMJr, 2015-06-15:
Thank You Will!

Ray

Jibeji, 2015-06-15:
Hi,
Tried the latest release," Return to Zero" goes on Machine's Zero position, not to the position defined by "Reset Zero".

RayMJr, 2015-06-15:
That is what I requested, above.

This would be why it would be better to have a user settable configuration option in UGS.

Ray

commit 1339d4fb (2015-06-14): Fix return to zero. Closes #258