winder/Universal-G-Code-Sender #261: UGS 2.0

tapnair, 2015-06-09:
I had 2 issues trying the nightly build, UGS 2.0.  I preface this with the fact that I have absolutely no idea what I am doing but jsut wanted to test it out.  Mainly because visualize wasn't working for me on 1.8 but making the change below visualize did work for me in 2.0 on Mac OSX 

1. It wouldn't compile because of an issue in GUIBackedn.java
I replaced:
         systemStateBean.setMachineX(Utils.formatter.format(this.machineCoord.getX()));
         systemStateBean.setMachineY(Utils.formatter.format(this.machineCoord.getY()));
         systemStateBean.setMachineZ(Utils.formatter.format(this.machineCoord.getZ()));
With:
        systemStateBean.setMachineX(Utils.formatter.format(this.machineCoord.x));
        systemStateBean.setMachineY(Utils.formatter.format(this.machineCoord.y));
        systemStateBean.setMachineZ(Utils.formatter.format(this.machineCoord.z));

It looks like the getX() methods are available on javafx.geometry.Point3D objects not javax.vecmath.Point3d

Am I missing something maybe?

2. Serial ports not working?  When I launch 2.0 serial ports are not recognized.  I get a popup that says: "No serial ports found."  I have created /var/lock with 777 permissions and serial ports are working with version 1.8

winder, 2015-06-10:
I'm really not sure why you get the build error. I would guess that there is something in the IDE which isn't configured quite right. What happens when you run "ant onejar-build-release" from the root of the project?

There is a new serial library in 2.0 called "JSSC", its supposed to come bundled with all the native requirements including OSX. Maybe it's just the port discovery which is broken, are you able to type in your serial port and connect?

redonebe, 2015-07-15:
To fix the getX() error I had to remove vecmath.jar from System/Library/Java/Extensions. This was an old version 1.3.

redonebe, 2015-07-17:
The problem with the serial ports is because the call SerialPortList.getPortNames() looks for these ports: file.getName().matches("tty.(serial.*|usbserial.*)"). I created a symbolic link for the GRBL port and now the port is showing in the dropdown. 

My serial port was  /dev/tty.wchusbserial14130 so i created the link like this: ln -s /dev/tty.wchusbserial14130 "/dev/tty.serial.GRBL"

georgeHayRiv, 2016-01-05:
I'm not all that good with regex expressions.
If the code just wants to narrow down the list to tty serial ports I think the regex expression could be simplified to this 
"tty.(.*?serial.*.)"
It looks for "tty."
then for zero or more characters not being greedy with the ".*?"
then for "serial" with "serial"
then for zero or more characters being greedy with the ".*"
then the "."
The parentheses are optional so 
"tty..*?serial.*."
This works for 
/dev/tty.wchusbserial14130
/dev/tty.serial.GRBL
/dev/tty.usbserial.GRBL

I can also create a symbolic link but getting the code changed might make it easier for the next guy.

redonebe, 2016-01-05:
This is not an UGS issue but a problem with SerialPortList. My advice is to buy a genuine Arduino and not a fake one with the 340 chip.

winder, 2016-01-05:
@georgeHayRiv I agree, this is code in the serial library though. It looks like there's an open issue there: https://github.com/scream3r/java-simple-serial-connector/issues/84

winder, 2016-01-05:
You can manually type in /dev/tty.wchusbserial14130 into the serial port input box, UGS will remember that port the next time you launch the application.

georgeHayRiv, 2016-01-05:
I didn't realize that you could type into that field. That works.
I did end up buying a Ramps board and LCD display. It is more than what I
need but if I can get that working I should be able to run off of a SD card.

George Adams
Hay River Software LLC
N14397 380th St.
Ridgeland WI 54763
715 455-1652

closed 2016-01-12 by winder