winder/Universal-G-Code-Sender #262: "Visualize" Don´t work on RPI with Debian Wheezy

elmarnitsche, 2015-06-15:
Hi,
on my mac all is fine, but on Raspberry 2 with Standard Debian Wheezy the "Visualize" button don't work. 

BR Elmar

elmarnitsche, 2015-06-15:
Ok1
got this on console:

Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: Could not initialize class javax.media.opengl.GLProfile
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:263)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:230)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:217)
	at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.<init>(VisualizerCanvas.java:135)
	at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.<init>(VisualizerWindow.java:71)
	at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1666)
	at com.willwinder.universalgcodesender.MainWindow.access$1500(MainWindow.java:80)
	at com.willwinder.universalgcodesender.MainWindow$11.actionPerformed(MainWindow.java:506)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6527)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)
	at java.awt.Component.processEvent(Component.java:6292)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4883)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4705)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4705)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

winder, 2016-01-12:
Duplicate #47

closed 2016-01-12 by winder