winder/Universal-G-Code-Sender #265: Downloaded, but no Universal-G-Code-Sender.jar file in the root directory

OldSalt1945, 2015-06-20:
I am new to all this, so it may be something of an "ID 10 T" problem (squish the letters together, spells IDIOT) , but I followed the directions found on youtube from NYCCNC, downloaded the zip file, expanded and saved it on my Win 8.1 laptop.

I could not find the .jar file neither on the expanded nor the zipped folders.

Is there some little secret I'm missing.

Thanks for any help.    John

winder, 2015-06-20:
See the download link in the readme not the source code download link for
github.

OldSalt1945, 2015-06-20:
Much better!  Helps when you do it right!

Thanks for the help.  Now all I have to do is figure out how to download Java something or other.

Thanks again.    John

closed 2015-06-22 by winder