winder/Universal-G-Code-Sender #267: Probing & Auto levelling

Northernboy, 2015-06-27:
Would be nice to have probing & auto levelling on the Z axis for milling PCB. 
Same as what Chillipepper does.

winder, 2015-06-27:
It's on the roadmap. Once I have the UGS Platform running with all the features the classic GUI has I'm going to start adding modules that do things like this.

winder, 2017-04-18:
It's probably a little early to share this since there is still a lot more work to do (and my machine isn't even setup to test it), but I'm really pleased with how well the UI proof of concept turned out. This is the culmination of the last 1.5 years of UGS Platform development to allow these kinds of features to be implemented as standalone plugins. The bottom panel with all the settings is a standard NetBeans plugin which registers the red previewer as a plugin for my visualizer.

The next steps to get a working prototype:
1. Generate the surface grid by sending a series of probe commands (recently added to the UGS core library).
2. Apply the surface data to the gcode file with a new plugin for the gcode parsing system (which was implemented early last year). I already have a gcode parser.

[image]

vMeph, 2017-04-18:
Wow awsome news cant wait to test it out😀

DOUG888, 2017-04-18:
Looking good Will, thanks for the update…as always appreciate all the effort You put in.

Doug

 

 <https://cloud.githubusercontent.com/assets/125509/25132838/d8218056-2418-11e7-905d-50e45e234b8c.gif> surface-scan-preview2

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Radiu, 2017-04-19:
@winder 
amazing :)
when you think will be possible share for we can test  has a autolevel helper test to help you with it?

winder, 2017-05-07:
This is still a work in progress, but I've been chipping away at the work and am really happy with the progress. The most complicated new pieces around UX, visualizer integration and the probe cycle, are all working nicely.

What remains is to slice the gcode into line segments which can be and adjust the z values according to the surface mesh. The most challenging part is slicing arcs. Fortunately, the UGS gcode parser has had this feature since the GRBL 0.8 days when arc performance wasn't as good as it is today.

If all goes well it should be available for testing in another couple weeks.

[image]

vMeph, 2017-05-07:
just amazing how the progress is going, looks very nice, looks promising
ROX!!!!

thanks alot @winder for all the work you have been providing for community, you the man!!!

thank you :)

winder, 2017-06-07:
Continuing to make progress on this feature. Summer has been pretty busy so far so I'm chipping away as I have time. I've also been working on the hardware side, and now have a machine with homing switches (but still need to make a probe attachment):

Last night I made a breakthrough and was able to get all the different pieces working together using test data:

[image]

[image]

There are some visualizer tweaks I'd like to add. Specifically, things are getting too busy so it's hard to see the gcode program. I'll probably add some options to the right-click menu to show/hide different graphical elements so I can disable different components (like the axis lines, probe grid, etc).

vMeph, 2017-06-07:
Looking a amazing just cant wait to check it out,any idea when a beta test is ready to test it out?

winder, 2017-06-11:
Implemented with #669

closed 2017-06-11 by winder

vMeph, 2017-06-13:
@winder do you think it would be hard to implement on UGS platform to accept .brd files format from pcb eagle like chilippr, for PCB milling, that would rox

winder, 2017-06-13:
@vMeph I doubt I'll take on that project.

vMeph, 2017-06-13:
@winder i wasnt able yet to figuer out how the autolevel works could add like a legend of what to expect from all the buttons and settings, that way would be better understanding?

on the settings button, when you click you see then the probe feedrate square dont matter how you change that will not alter the velocity that the Z feedrate moves is always the same if you put 10 or 200 the speed of Z is always same
also on settings when you change values and you click apply that will make all changes to go to default

this squares would be great to be allign
[image]

vMeph, 2017-06-13:
also notice that when is doing the autolevel when it touches plate and goes to other dot doesn't raise Z first, it goes to the other dot and raising at same time,at least thats what im seing it, i think the expectation would be to raise first and only then move to the next dot

winder, 2017-06-13:
Thanks for the feedback. I'll fix these 2 bugs and see about some
documentation soon.

winder, 2017-06-14:
@vMeph I fixed the settings page and configured it to raise the tool between probes. The GUI is sort of temporary right now. It looks good on my machine but I used a drag and drop GUI builder to put it together quickly but things always seem to look different on linux/mac/windows with that tool.

The closest thing to documentation right now is here:
https://github.com/winder/Universal-G-Code-Sender/pull/669

When you say "would be great to create a border line on outside dimentions and inside the matrix dots" are you talking about the outside dimensions of your gcode program?

There are a couple other unrelated bugs I need to fix before I can test this on my machine, so I'm sure you'll find other issues. Thanks for helping out.

vMeph, 2017-06-14:
@winder 

border line on outside dimentions and inside the matrix dots i mean something  like this
[image]

the build #774 and #775 failed

thanks

vMeph, 2017-06-15:
@winder 
i have tested today and now for the little test i did it seems that autolevel is doing what is suppose to do, thanks for the border line.

Now there is something that isn't working as expected, after make all the probe areas i view the data and apply the Ggode, and then i click on send button on top off the window, and what is happening is machine will start to moveto do the  job, but will not turn the drill on that is on the original file , it seems that when you apply the Gcode it will ignore some of the original code, i have tested with a small simple circle
 i would think when you apply the G code the new data of autolevel it should only apply changes into the Z areas of the code?
this is the circle code i used
%
(1001)
(T12  D=1.4 CR=0 - ZMIN=-1 - flat end mill)
G90 G94
G17
G21
G28 G91 Z0
G90

(2D Contour1)
M9
T12 M6
Z10
G54
M8
G0 X13.86 Y29.12
S1000 M3
Z5
G1 Z2 F200
Z-0.86
G19 G2 Y28.98 Z-1 J-0.14
G1 Y28.84
G17 G3 X14 Y28.7 I0.14
G2 Y-0.7 J-14.7
Y28.7 J14.7
G3 X14.14 Y28.84 J0.14
G1 Y28.98
G19 G3 Y29.12 Z-0.86 K0.14
G0 Z10
S0 M5
G17
M9
G28 G91 Z0
G28 X0 Y0
M30
%

vMeph, 2017-06-15:
@winder  Apply to Gcode button also have a bug, when you click the button "apply to Gcode" will add data but if you press more times the button it will keep adding, i would think it should only apply to it only 1 time the Gcode data, if you try it out you will see in visualizer that  every time you click apply to gcode button it will keep adding more and more data

winder, 2017-06-15:
@vMeph there are some other edge cases too, if you click it before doing the probes I think there's an error.

vMeph, 2017-06-22:
@winder any news why after doing the autolevel and apply the data and send  play to do the job it will ignore some of the original job file like the S1000 M3 as example ,last time i check the drill wouldn't turn on after applying the data of the autolevel

winder, 2017-06-25:
@vMeph I found some problems using the test program you provided. I think they're resolved now.

vMeph, 2017-06-25:
@winder i just did a small test with code above and spindle doesn't turn on still
if i use normal with out using autolevel, works normal
if i use with autolevel and apply then data spindle dont turns on
it seems the problem is when you apply the new data , must be something wrong when you apply the new data will override the original code in parts that shouldnt do

Radiu, 2017-07-02:
@winder any news on autolevel im having same problem has @vMeph report

winder, 2017-07-04:
@vMeph @Radiu I found a couple issues in the gcode parser while debugging this problem. I've got a fix but need to do some more testing.

vMeph, 2017-07-04:
@winder while you improve if you keep adding updates to the builts i will debug with my machine if it helps

winder, 2017-07-04:
@vMeph thanks, there were some obvious problems remaining the last time I had a chance to work on this otherwise I would have.

winder, 2017-07-08:
@vMeph @Radiu I just updated the nightly build with some changes that should fix this problem. The issue had to do with how I handled implicit gcode commands, when you write something like this the second command should be treated as `"G1 X2 Y3"` that was working fine:
```
G1 X1 Y2
X2 Y3
```

The problem was when you have a non-movement command like this:
```
G0 X13.86 Y29.12
S1000 M3
```

In this case `"S1000 M3"` was being treated as though it were `"G0 S1000 M3"` which caused a strange chain of events leading to the `"S1000 M3"` being replaced with a `"G0"`.

vMeph, 2017-07-08:
@winder ok the spindle issue is fixed, but i wasnt able to make it work, not sure if is something im doing wrong or if is from software, if you could explain me a sequence of test, some options im not sure what they for yet, lets say i put a job that will make a 0.5 deep cut on a plate, the settings im doing is

i press use loaded file (the Z min and max detects file settings in this case Zmin0.5  Zmax10)
then i will change the Z min to like 4 and zmax to 5 cause if i leave what use loaded file have detected then will stop when Zmin gets to the 0.5

i set my resolution and then i hit scan surface ( probing is done with sucess)

then in view data and press apply to Gcode i notice the button blinks like is loading file or something,but i assume that when you press apply Gcode is instantanious applyed

other settings i did nothing i only have put feedrate values

when i start the job for some reason it will start milling above the material, i have testes with out the autolevel i start with drill bit on top of plate,start job and mills normaly

is there something im missing or some bug still on?

also when you hit the stop button all panels are greyed out ,so means if im starting a job and i decide to stop cause i missed something then i cant return to 0 position,ill be forced to disconect and conect machine to have acess to the common actions panel

vMeph, 2017-08-14:
@winder any news on autolevel progress?
i wasnt able yet to make it work proper or im doing settings wrong or not getting what im exepecting, any help in how to set up settings with steps?
also sometimes i get a error and a pop up comes saying (error9) G code commands are locked out during alarm or jog state              when this pop up comes you cant close it , you are force to hit control+ alt+delete to shutdown UGS program

vMeph, 2017-08-16:
@winder 
[image]

lets say you click use loaded file you have machine point 0 above the surface at 6 mm step if you start probing sense the Z is at -05 will cause this error to pop up witch is ok, the problem is that eaven if you click ok you cant close window the only solution is to control+alt+delete to terminate UGS program

also i think i have located the problem but im not sure what is causing
its located on the use loaded file button action
if i dont use the button "use loaded file " and do all X Y Z size manualy all worked has expected, but when i use the button use loaded file the Z always mills above the surface
maybe is inverted the direction?

winder, 2017-08-16:
@vMeph I haven't had much time to work on this recently. I'd like to switch back to the Z-Probing for a bit and make sure that workflow is solid and easier to use (looking at some new GUI tools), then take what I learn from that and take another stab at the auto-leveling workflow. I'm thinking something along the lines of a "wizard" style interface that walks you through each step.

For the error you found, the problem is that the surface probing algorithm isn't very smart and keeps sending commands even though an error has occurred. So there are a whole bunch of error messages stacked on top of each other.

vegasguitars, 2017-08-18:
How are the Z-probing efforts coming along?

Poly3D, 2017-10-24:
I just got my probe up and running on my Carvey and after probing blank PCB successfully the resulting probing grid looks a little exaggerated, when I apply it to the g-code I and run it, I can see Z axis vary by approximately 2mm, just like the Visualizer shows. The PCB definitely isn't warped this much. 
Is it something that you are still working on? Or am I doing something wrong? 
Great software though!

wasanka, 2018-01-05:
Hi, i got the auto level working with a probe. Once all the points are probed and I click on view data button, it does not have data for all the points probed.

pulss, 2018-02-03:
Hello,
I have the same problem, after scanning, there are no data.
Paul

eduardogi, 2018-02-04:
Hello
I have the same problem. I'm using the version 201609300101.
Regads

njrl, 2018-06-06:
same problem 
201609300101 nightly june 4 2018
got anybody solved that problem 
regards

luisluna2, 2018-07-29:
YES, NOT WORK NOW.

BEST TRY WITH GRBLCONTROL: 

[https://github.com/Andypos/grblControl](url)
[https://www.youtube.com/watch?v=tK5Amg-Ye58](url)

hoangtrthien, 2018-11-10:
Hi.
I have the same problem. No data after scanning. I am using Universal Gcode Platform 2.0 Nightly/ Nov 05, 2018
Has anyone solved this?

Regards

Joe-Soap, 2019-08-24:
Please stop Autoleveler freaking out and popping up 70 million Error message boxes when something fails with the probe

butterflyam, 2025-04-11:
非常好