winder/Universal-G-Code-Sender #270: Some proposals for the GUI

BJDayzz, 2015-06-30:
I'm not using the program for a long time since I'm quite new in CNC machining. But so far I'm really happy with it. Only a few cosmetical things could be improved from my point of view:

- After a job is completed, the "Send" button in "file mode" tab should be enabled again. (Currently the button is disabled after a job is done and I have to reload the file if I want to run it again.)
- The "Reset to Zero" button should be bigger or marked in a different color. (So often I press "Return to zero" instead of reset.  :-(   )
- The selected unit (inch / mm) under "machine control" tab should be stored for the next program start. Currently the setting "mm" is changed to "inch" whenever the program is closed and restarted.
- The keyboard movement option should be disabled while the cursor is in the step size field. (When I e.g. just want to enter a 4 in the field with enabled keyboard movement, my machine runs in -x direction.)

Would be great if you could realize some of these proposals - or give me a hint if I'm just doing something wrong.

Thank you!

winder, 2015-06-30:
Thanks for the suggestions.

For your first issue, try using the most recent 2.0 release. The sent button is supposed to be re-enabled after a file is done being sent. If it is still an issue there I can investigate further.

BJDayzz, 2015-06-30:
Wow, that was fast! :-)

Yes, I was using 1.0.8 before. Now I tried 2.0 but it doesn't start. Only
the info window "..nightly built.." appears. For both versions I use the
windows batch file. Do you have any idea what the problem could be?

BJDayzz, 2015-06-30:
Here's the log file:

[JarClassLoader] INFO:
 setProperties(com.simontuffs.onejar.JarClassLoader@60e53b93)
[JarClassLoader] INFO:
 oneJarPath=file:/C:/_sell%20und%20jenes/GRBL/UGCS_2_0/UniversalGcodeSender.jar
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:
 oneJarPath=file:/C:/_sell%20und%20jenes/GRBL/UGCS_2_0/UniversalGcodeSender.jar
[JarClassLoader] INFO:  findResource() found:
"com/willwinder/universalgcodesender/utils/Version.class" for caller null
in codebase main/Universal G-Code Sender.jar
[JarClassLoader] INFO:  findResource(): unable to locate
"resources/MessagesBundle_en.properties"
Jul 01, 2015 12:26:38 AM
com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFORMATION: Loading settings.
Jul 01, 2015 12:26:38 AM
com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFORMATION: Settings file location: C:\Users\BJ\UniversalGcodeSender.json
Jul 01, 2015 12:26:39 AM
com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFORMATION: Applying settings.
[JarClassLoader] INFO:  findResource() found:
"libs/windows/jSSC-2.8_x86_64.dll" for caller null in codebase lib/jssc.jar
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at com.simontuffs.onejar.Boot.run(Boot.java:315)
        at com.simontuffs.onejar.Boot.main(Boot.java:152)
Caused by: java.lang.UnsatisfiedLinkError:
jssc.SerialNativeInterface.getSerialPortNames()[Ljava/lang/String;
        at jssc.SerialNativeInterface.getSerialPortNames(Native Method)
        at jssc.SerialPortList.getWindowsPortNames(SerialPortList.java:305)
        at jssc.SerialPortList.getPortNames(SerialPortList.java:294)
        at jssc.SerialPortList.getPortNames(SerialPortList.java:178)
        at
com.willwinder.universalgcodesender.utils.CommUtils.getSerialPortList(CommUtils.java:44)
        at
com.willwinder.universalgcodesender.MainWindow.loadPortSelector(MainWindow.java:2396)
        at
com.willwinder.universalgcodesender.MainWindow.initProgram(MainWindow.java:2098)
        at
com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:122)
        at
com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:204)
        ... 6 more

winder, 2015-07-01:
It looks like this might be an open issue with the serial library I'm using:

https://github.com/scream3r/java-simple-serial-connector/issues/72

Does your home directory have permissions set to readonly somehow?

winder, 2016-03-05:
I think most of these are fixed now, except for cleaning up the "Reset to Zero" and other buttons. I'm hoping to get to that soonish while making "Workspace Coordinate System" management better.

closed 2016-03-05 by winder