winder/Universal-G-Code-Sender #271: Visualize not working

Bigfellas, 2015-07-03:
Hi I am runnung the latest nightly 2015-06-29. 
I am running it on a raspberry pi v2. But I can not get the Visualize to work. 
Thanks Matt

rodcar125, 2015-08-28:
Dont know what date im running of the nightly build, I downloaded it yesterday 08/27/15 and im also having problems with the visualizer, click the button but nothing happens. any suggestions?

georgeHayRiv, 2015-11-06:
I am using OSX 10.11.1 with java version "1.8.0_25"
 Visualize is not working with Universal-G-Code-Sender version 1.0.8
Here is the console read out after clicking the Visualize button.

Exception in thread "AWT-EventQueue-0" java.lang.NoClassDefFoundError: Could not initialize class javax.media.opengl.GLProfile
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:263)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:230)
	at javax.media.opengl.awt.GLCanvas.<init>(GLCanvas.java:217)
	at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.<init>(VisualizerCanvas.java:131)....

I did need to download the librxtxSerial.jnilib from http://blog.iharder.net/2009/08/18/rxtx-java-6-and-librxtxserial-jnilib-on-intel-mac-os-x/ to even get the application to run. RXTX is still being used for something.
I also tried to download the source and work with it in Eclipse. There are 106 errors mostly in VisualizerCanvas.java. A number are import errors like this
import static com.jogamp.opengl 
I think I got all the jodamp .jar files added to the classpath.

splitn2, 2015-11-06:
I've got the same problem on RPi , I'm using 2.0 nightly build just downloaded a few nights ago, and latest Jesse Raspian image. Otherwise UGS is working brilliantly and I am really impressed. But I would love to have the visualiser. cheers

winder, 2015-11-06:
I believe that this version of jogl only works on RPi with the "newt"
canvas, I was never able to get it ported over at the time.

In the 2.0 platform code I got the visualizer running with the latest
version of jogl, but never back ported it to the classic GUI.

If you try the UGS platform build does the visualizer work?

splitn2, 2015-11-06:
@winder , just wondering ..... to install and run the 2.0, is it just a matter of downloading and extracting the zip, then running one of the binaries in the bin directory ie. for RPi execute this ...... ./ugsplatform/bin/ugsplatform ??? appreciate your help I am looking forward to using this, it might be put to work on my much larger 2.55m x 1.3m x 250mm cutting area router soon too!

winder, 2015-11-06:
Yeah, just run one of the executables for your platform. Let me know how it
goes.

splitn2, 2015-11-06:
Thanks @winder I'll try the 2.0 Platform tonight, cheers

georgeHayRiv, 2015-11-06:
Visualize is working for me on Mac OS X 10.11.1 El Capitan running the executable located at ugsplatform/bin/ugsplatform.
I never would have guessed that was how I would get that to work. You may want to add some instructions on the web site or in the download folder.

winder, 2015-11-06:
@georgeHayRiv The visualizer in the classic GUI should be fine on any version of OSX in the latest build.

georgeHayRiv, 2015-11-06:
Visualize in the classic GUI is also working.

winder, 2016-02-17:
Duplicate of #47

closed 2016-02-17 by winder