winder/Universal-G-Code-Sender #273: Macros not saved...

Crazydoub, 2015-07-05:
Macros are not saved so it must be re-write each appli startup :/

winder, 2015-07-05:
What version are you using?

How do you exit the program?

Crazydoub, 2015-07-05:
I use master version Nightly 2.0 :D
I exit with window Cross.

HarryT36, 2015-07-05:
Even if you change the macros directly in the *.json file they are not used in UGS 2.0 nightly built

HarryT36, 2015-07-05:
I must recall my last post. Thee reading is working now.

Crazydoub, 2015-07-05:
In fact, Macro are saved using [Enter] key...
When you know that, all works!

Crazydoub, 2015-07-05:
Just need to Add Start/Stop Spindle to Machine Control tab and i definitively leave GRBL Controller To UGS :D
Great Job Will :+1:

HarryT36, 2015-07-05:
I also tested this and it works as descripted. The macro is saved when using [enter] as last "Character" in a command line. It only appears strange that the cursor remains in the used line and nothing seems to happen.

okolloen, 2015-08-02:
Wouldn't it be better to read the state of the macros when saving the preferences in the addShutdownHook method (about line 150 in MainWindow). 
Adding the lines : 
settings.setCustomGcode1(customGcodeText1.getText());
settings.setCustomGcode2(customGcodeText2.getText());
settings.setCustomGcode3(customGcodeText3.getText());
settings.setCustomGcode4(customGcodeText4.getText());
settings.setCustomGcode5(customGcodeText5.getText());

there would mean you do not rely how the user enters the macro.

winder, 2015-09-14:
Thanks @okolloen, you're right that was the problem.

commit 30e1e012 (2015-09-14): Save macros on shutdown. Closes #273
closed 2015-09-14 by winder