winder/Universal-G-Code-Sender #274: Machine Control-Return to Zero-Axis sequence incorrect

woodworkerbob, 2015-07-06:
Version 2.0 from 2 days ago. In the Machine Control Panel, Return to Zero button sends:
G90 G0 Z0
G90 G0 X0 Y0 Z0

This is bad as the router bit can drag across the material while returning to zero.

Suggestion:
G90 G0 X0 Y0
G90 G0 Z0

Better improvement.  If the machine has homing switches, move Z axis to encountering switch, move to X 0 Y 0, then to Z 0.

Bob

winder, 2015-07-06:
The first command should be "G90 G0 Z<MAX>", UGS watches commands as they
are processed and updates the MAX variable. So if you had previously homed
the machine I would expect max to be set differently. If you haven't moved
the machine yet it will use zero.

woodworkerbob, 2015-07-06:
I was reading from the console listing,  verbose mode off.  Let me run the process again with verbose on.  I have done this twice now and my process must be wrong as the router bit glides just touching the material surface.

woodworkerbob, 2015-07-16:
Will,
I'm searching for how to change the default behavior for a tool change 
command.  It seems right now the tool change is ignored.

I would like to suggest that when a tool change is encountered, UGS 
would print out the which tool is being requested and ask for user input.

Bob

winder, 2015-07-16:
Could you create a new ticket for this?

I'm not sure what GRBL does for a tool change. Is there a gcode supported
(or ignored) by GRBL? Does it trigger a feed hold?

One way you could do this with no changes to UGS is to put a comment above
or after your tool change command. UGS displays the most recent comment in
the status area.

chamnit, 2015-07-16:
@winder : Grbl gives an error upon an M6 command. It leaves it up to the user/GUI to intercept and handle these based on machine/GUI/user preferences, as they can be drastically different. Keep in mind that it's EXTREMELY important to record the g-code state and position before injecting any non-job commands into a program. You have to ensure after the GUI completes that the g-code state and position is returned to what it was. Otherwise, you can corrupt the g-code program from it's oringal intent.

jpeast123, 2016-07-16:
I am having this problem.  When I push return to zero, the z axis goes to zero then drags the bit on top of the material.  I would like z axis go to zero to be last in that chain.

jahnj0584, 2016-07-16:
I posted just that in issue #414 jp

closed 2016-07-31 by winder