winder/Universal-G-Code-Sender #278: Universal g-code sender and command line

ivanterzo, 2015-07-13:
In raspberry (and then Linux) is possible run the software directly from the command line by automatically open the serial port (USB), and load the desired file? If not, I think that would be an interesting feature to be added

winder, 2015-07-13:
It wouldn't be too hard to make this now that the GUI is fully decoupled
from the back end, I even made a prototype at one point (before the
decoupling). Flushing out the feature and adding it to the jar isn't high
on my TODO list though.

ivanterzo, 2015-07-13:
if you want I'll explain what I was thinking about this feature.
In practice the cnc I'm finishing has a raspberry (or a mini-ITX mobo with linux), a touchscreen and can be controlled from the Internet via VNC.
I would not want to give complete control of the program but, rather, a web access to a page written in php (or other language) where the user after providing their credentials and upload the file, "active" the software (the software start via php functions. The user does not see the interface of the program but it is loaded locally).
I hope I was clear enough with my poor English.
Thanks again for your attention

winder, 2016-01-19:
I don't see myself implementing this in the near future. UGS is now a reusable library, so if someone is inclined they could use it to create a CLI application.

closed 2016-01-19 by winder