winder/Universal-G-Code-Sender #279: Gui enhancements

Crazydoub, 2015-07-14:
[image]

Jibeji, 2015-07-15:
Bonjour,
Pretty good!
Just a suggestion, a caption for the button could be better than the Gcode.
As an example, one of my macro is:
G21;G91 G38.2 F20 Z-45; G92 Z+1.5; G91 G0 Z1.5 (Z-probe)
Not sure it will fit... ;)

Crazydoub, 2015-07-15:
My scrit take the 18 first letters of the gcode ;) but caption would be great yes!

Envoyé par TypeMail

winder, 2015-09-14:
Great idea. There are utility functions in UGS to pull out just the comment, I think that would be perfect for the button text.

Do you have a public fork of UGS with these changes in it?

Crazydoub, 2015-09-14:
I got modified source biy dont understand how to post it on github...

Envoyé par TypeMail

closed 2016-04-10 by winder