winder/Universal-G-Code-Sender #280: send file gcode directly to serial port

ivanterzo, 2015-07-17:
On linux I'm trying to send a file gcode directly to serial port.
I open the serial port with:
	minicom -b 9600 -d / dev / ttyACM0
and from another console read line by line the file gcode and I send to serial:
	sed -n '1, /' * '/ p' <file_gcode>> / dev / ttyACM0
The problem is like that commands are executed up to a certain point, but then, nothing. As if it came at the end of the file.
You can also try if you want, you will see that to a certain point everything seems to be going well, but then, stop!
Suggestions?
thanks

winder, 2015-07-17:
You aren't sending the file correctly. GRBL has a limited buffer and you shouldn't send more data than can be handled by that buffer.

The protocol is documented here:
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-a-g-code-program-to-grbl

Simple CLI scripts are available here:
https://github.com/grbl/grbl/tree/master/doc/script

closed 2015-07-17 by winder

ivanterzo, 2015-07-17:
I was in fact to add what appeared to be a problem of an excess of data sent.
Thanks, thanks and thanks again :)