winder/Universal-G-Code-Sender #282: Bug in visualizing helical motion

DaveBrauns, 2015-07-18:
There seems to be a bug in the visualizer when using G02/G03 together with movement in Z-direction, on end of motion is a step displayed. See image
[image]

winder, 2015-07-19:
Thanks, there's probably an off-by-one error in the arc to line segment algorithm.

Could you share the Gcode where you're seeing this?

DaveBrauns, 2015-07-19:
Of course

G00Z5.000
G00X47.000Y100.500
G00Z0.500
G02X46.000Y99.634Z0.000I-1.000J0.000
G02Y101.366Z-0.500I0.500J0.866
G02X47.500Y100.500Z-1.000I0.500J-0.866
G02X46.000Y99.634Z-1.500I-1.000J0.000
G02Y101.366Z-2.000I0.500J0.866
G02X47.500Y100.500Z-2.500I0.500J-0.866
G02X46.000Y99.634Z-3.000I-1.000J0.000
G02Y101.366Z-3.500I0.500J0.866
G02X47.500Y100.500Z-4.000I0.500J-0.866
G00Z5.000

DanielWeigl, 2019-04-29:
this is most likely fixed in https://github.com/winder/Universal-G-Code-Sender/pull/1220 / https://github.com/winder/Universal-G-Code-Sender/pull/1220/commits/5dd39fec77589e7d572ebcf6fcb47abe00fede84

closed 2019-04-29 by winder