winder/Universal-G-Code-Sender #283: Raspberry Pi issues with UGCS 2.0

neckyeskia, 2015-07-28:
Hello, 
I recently installed 2.0 on a Wheezy OS, but I'm having problems getting response from my machine. I get an error message indicating that "some or all services may not be available" or something to that effect, along with a list of a number of features. Wondering if someone smarter then I can help me out? What am I missing?
Thanks!

winder, 2015-07-28:
Could you be more specific about the error you're getting?

For instance is the error happen when launching the program, or opening the connection, or sending a gcode file?

Also what is the exact error message? I couldn't find any reference in the code for "some or all services may not be available", nor could I find any on the google related to the Raspberry Pi.

neckyeskia, 2015-07-28:
Sure. This is happening when I go to open a connection. See photo of exact
error message.

Thanks.

mattbecka, 2015-08-14:
Any love for this? Still struggling.

winder, 2015-08-14:
If you're using the same version as in the screenshot (June 29th build of 2.0), try downloading the most recent nightly build. There were a lot of changes to this part of the code in July.

winder, 2015-09-14:
Is anyone still having this issue with the latest release?

closed 2016-01-19 by winder