winder/Universal-G-Code-Sender #285: <ldl

Spica2, 2015-08-02:
The Console returns
    Grbl 0.8c ['$' for help]
    <ldl
again and again.
Arduinos Monitor doesnt.
Should be the UGS?
Environment Arduino Uno and a Mac.

guru54321, 2015-09-24:
I've seen similar behaviour on my Grbl when the eeprom contains funky values. After clearing the eeprom (see https://www.arduino.cc/en/Tutorial/EEPROMClear)  and reinstalling grbl on my arduino, this issue solved itself. In my case, the eeprom got corrupted due to me experimenting with a 4 linear axis grbl I build myself. I would not know why this could happen in your case though. Normally grbl doesn't do this.. (not to my experience at least, but then, I just start using it)

Spica2, 2015-09-25:
Thanx for your reply. But do not get a solution. Maybe I do it wrong as I am new? Put the EEPROM Clear Code in the IDE, saved, put the EEPROM.h in the new folder and uploaded. Now I get 
Grbl 0.8c ['$' for help]
<ldle,MPo
again and again

winder, 2015-09-25:
If you downgrade then upgrade GRBL it should also fix the problem.

Spica2, 2015-09-28:
Uh, also have problems with the hexuploader. Does not work/upload. Already reportet at its repository. Any other suggestion meanwhile I am looking for a hexuploader fix?

guru54321, 2015-09-28:
Did you check on the baudrate? My nano only uploads at 56Kbit although 112 works fine when communicating with grbl...

Spica2, 2015-09-29:
Checked all baudrates! No sucess.

Spica2, 2015-09-29:
ok. now cleard eeprom and flashed grbl 9 from inventables (previous 8 from other someone else) by ide. now runs as expected.

winder, 2015-09-29:
Great, thanks @Spica2 and @guru54321

closed 2015-09-29 by winder