winder/Universal-G-Code-Sender #288: Configuration file suggestion for Linux

Crispy24, 2015-08-13:
Configuration files for Linux systems should be located in the following locations:

    ~/.UniversalGcodeSender.json
    /etc/UniversalGcodeSender.json
    /usr/share/UniversalGcodeSender/UniversalGcodeSender.json

(Please note the period prepended in `~/.UniversalGcodeSender.json`)

Configuration files should also be sourced in that given order, too.

Another reasonable approach would be to implement a command-line option for specifying the configuration file's location. Something like `$ ugs -c ~/.custom-config.json` or `$ ugs --config="~/.custom-config.json"`. This would tie in with issue #278.

Or for the best functionality: combine the two. Implement a config sourcing order and a command-line option. Linux users, package maintainers, and developpers wanting to use UGS in their scripts and projects will benefit greatly.

winder, 2015-09-14:
Thanks for the suggestion, I'd like to add something like this.

winder, 2016-02-20:
I need to write to the json file, the files in /etc and /usr/share are owned by root. Do you know how permissions on files in there normally work?

jstamant, 2016-02-20:
Good point. I forgot to consider that UGS writes to the json file at run-time. You don't want UGS to write to the files I mentioned, they're static files in the Linux FHS. Although `/etc` and `/usr/share` remain the perfect place to store a default, *non-editable* config.

e.x. You'll typically see programs copy a fresh config file from `/usr/share` to the user's home directory.

At the very least, what I'd love for you to take from this suggestion is that the period should come before the filename like I suggested: `~/.UniversalGcodeSender.json`

That's what differentiates a Linux user's personal documents from his configuration files.

winder, 2016-02-20:
Done! The default settings file is hidden now. I'm still checking for hidden and visible files though, so you'll need to rename it to be hidden to cleanup your home directory.

commit dc1c1d67 (2016-02-20): Only check Linux home.dir for property file, default to a hidden file if no file is found. Closes #288
closed 2016-02-20 by winder