winder/Universal-G-Code-Sender #290: Winder - Can I trouble you for help getting the UGCS onto a raspi?

MandEdesign, 2015-08-28:
Hi, 

Having a bit of trouble getting the UGCS .jar file to execute on raspi - Wheezy
using openjdk 7 

Here was my last attempt at the terminal trying to execute the .jar
___________________________________________
login as:
login as: pi
pi@192.168.0.2's password:
Linux raspberrypi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug 17 23:38:05 2015 from whip.local
pi@raspberrypi ~ $ cd Desktop/UGCS
pi@raspberrypi ~/Desktop/UGCS $ java -jar UniversalGcodeSender.jar
[Boot] INFO:  myJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSender.jar
[JarClassLoader] INFO:  setProperties(com.simontuffs.onejar.JarClassLoader@cbb61
[JarClassLoader] INFO:  oneJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSend
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:  oneJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSend
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.                                                                                                                                                             java:57)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAcces                                                                                                                                                             sorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:606)
        at com.simontuffs.onejar.Boot.run(Boot.java:315)
        at com.simontuffs.onejar.Boot.main(Boot.java:152)
Caused by: java.lang.InternalError: Can't connect to X11 window server using 'lo                                                                                                                                                             calhost:10.0' as the value of the DISPLAY variable.
        at sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)
        at sun.awt.X11GraphicsEnvironment.access$200(X11GraphicsEnvironment.java                                                                                                                                                             :65)
        at sun.awt.X11GraphicsEnvironment$1.run(X11GraphicsEnvironment.java:110)
        at java.security.AccessController.doPrivileged(Native Method)
        at sun.awt.X11GraphicsEnvironment.<clinit>(X11GraphicsEnvironment.java:7                                                                                                                                                             4)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Class.java:191)
        at java.awt.GraphicsEnvironment.createGE(GraphicsEnvironment.java:102)
        at java.awt.GraphicsEnvironment.getLocalGraphicsEnvironment(GraphicsEnvi                                                                                                                                                             ronment.java:81)
        at sun.awt.X11.XToolkit.<clinit>(XToolkit.java:120)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Class.java:191)
        at java.awt.Toolkit$2.run(Toolkit.java:869)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:861)
        at sun.swing.SwingUtilities2.getSystemMnemonicKeyMask(SwingUtilities2.ja                                                                                                                                                             va:1925)
        at javax.swing.plaf.basic.BasicLookAndFeel.initComponentDefaults(BasicLo                                                                                                                                                             okAndFeel.java:752)
        at javax.swing.plaf.metal.MetalLookAndFeel.initComponentDefaults(MetalLo                                                                                                                                                             okAndFeel.java:434)
        at javax.swing.plaf.basic.BasicLookAndFeel.getDefaults(BasicLookAndFeel.                                                                                                                                                             java:148)
        at javax.swing.plaf.metal.MetalLookAndFeel.getDefaults(MetalLookAndFeel.                                                                                                                                                             java:1589)
        at javax.swing.UIManager.setLookAndFeel(UIManager.java:536)
        at javax.swing.UIManager.setLookAndFeel(UIManager.java:576)
        at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1345)
        at javax.swing.UIManager.initialize(UIManager.java:1455)
        at javax.swing.UIManager.maybeInitialize(UIManager.java:1422)
        at javax.swing.UIManager.getInstalledLookAndFeels(UIManager.java:416)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:1                                                                                                                                                             945)
        ... 6 more
pi@raspberrypi ~/Desktop/UGCS $ uname -a
Linux raspberrypi 3.18.11+ #781 PREEMPT Tue Apr 21 18:02:18 BST 2015 armv6l GNU/Linux
pi@raspberrypi ~/Desktop/UGCS $ uname -r
3.18.11+
pi@raspberrypi ~/Desktop/UGCS $
_______________________________________________

Also ran         java -jar -Xmx256m UniversalGcodeSender.jar and the same result
any suggestions?  Much appreciated in advance :)

MandEdesign, 2015-08-28:
Would it be possible to share an image where its pre - configured?

winder, 2015-08-28:
I haven't tried open-jre, you might try the oracle JRE

MandEdesign, 2015-08-30:
Hey so I removed openjdk and now have:

pi@raspberrypi:~$ java -version
java version "1.8.0"
Java(TM) SE Runtime Environment (build 1.8.0-b132)
Java HotSpot(TM) Client VM (build 25.0-b70, mixed mode)

Got a new message when I run the jar via 
java -jar UniversalGcodeSender.jar   or java -jar Xxmx256x UniversalGcodeSender.jar

pi@raspberrypi ~/Desktop/UGCS $ java -jar -Xmx256m UniversalGcodeSender.jar
[Boot] INFO:  myJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSender.jar
[JarClassLoader] INFO:  setProperties(com.simontuffs.onejar.JarClassLoader@e53b93)
[JarClassLoader] INFO:  oneJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSender.jar
[Boot] INFO:  using JarClassLoader: com.simontuffs.onejar.JarClassLoader
[JarClassLoader] INFO:  oneJarPath=file:/home/pi/Desktop/UGCS/UniversalGcodeSender.jar
[JarClassLoader] INFO:  findResource(): unable to locate "resources/MessagesBundle_en.properties"
Aug 30, 2015 10:08:33 PM com.willwinder.universalgcodesender.SettingsFactory loadSettings
INFO: Loading settings.
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:483)
        at com.simontuffs.onejar.Boot.run(Boot.java:315)
        at com.simontuffs.onejar.Boot.main(Boot.java:152)
Caused by: java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
        at java.lang.Runtime.loadLibrary0(Runtime.java:870)
        at java.lang.System.loadLibrary(System.java:1119)
        at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:83)
        at com.willwinder.universalgcodesender.CommUtils.getSerialPortList(CommUtils.java:45)
        at com.willwinder.universalgcodesender.MainWindow.loadPortSelector(MainWindow.java:2461)
        at com.willwinder.universalgcodesender.MainWindow.initProgram(MainWindow.java:2061)
        at com.willwinder.universalgcodesender.MainWindow.<init>(MainWindow.java:84)
        at com.willwinder.universalgcodesender.MainWindow.main(MainWindow.java:1963)
        ... 6 more

From what I can tell from browsing the .jar file "resources/MessagesBundle_en.properties" might actually need to be resources/MessagesBundle_en_US.properties" which I saw in the resources folder when browsing from xarchiver 

Thanks again for your help 

Matt and Toby

MandEdesign, 2015-09-02:
I got some help from a friend - Daryl Stultz

So........... 
1.Please disregard that first bit of traceback on Aug 27th as that was through putty (novice).  
2.We were able to execute the program by using the following strings.

$sudo apt-get install librxtx-java
$java -jar -Djava.library.path=/usr/lib/jni UniversalGcodeSender.jar   

I hope that this might help towards future dev.  Looking forward to learning more so that I can contribute something useful. 

M

closed 2015-09-02 by MandEdesign