winder/Universal-G-Code-Sender #293: (Noob alert) UGS Return to Home/Reset Home Button Function.

Jamieip, 2015-09-21:
I've always had issues with returning to home in UGS, and I think i've found out why.

The Machine home is set during homing and booting of grbl. This cannot change.
The Reset Zero function resets the Work Position
The Return to Home function returns the machine to the Machine Home. 

So I have been assuming that I can pick a home position for my work and then return to it with the button, this is not the case as the Return to home returns to the machine home, not work home.

I would like the "Reset Zero" button re-labeled to "Reset Work Zero".
I would like to see another button called "Return to work zero" which would be a G90 X0 Y0 Z0 I think.

(the noob bit) I have no programming experience except for the tutorials that come with Arduino.
What I'd also like to know is how to add my own buttons to UGS. I downloaded the files and installed JAVA JDK and installed Mavern.

winder, 2015-09-21:
Thanks this is a great suggestion I think it will clear up a lot of confusion.

For customizing the GUI you can do that with Netbeans, I used a GUI builder for most of it so there is a very shallow learning curve.

Jamieip, 2015-09-21:
Would you be able to briefly explain how I would go about this? what GUI builder would I need, what code would I need to make the button give a G-code command and how would I compile it into a new jar?

winder, 2015-09-21:
You can also use the macro tab to make custom commands, that might be a better option.

Jamieip, 2015-09-21:
Yeah that was my back up plan lol. Been messing around in Netbean for a few hours and can't figure it out. I do have a few ideas that i'd like to implement in a build and show you but I think it's not gonna be as simple as I thought.

winder, 2015-09-21:
In Netbeans open "MainWindow.java", you should see the GUI builder pop up. Double click on a button and it will show you the code attached to it.

Jamieip, 2015-09-21:
yeah been around that. My issue is that I can't work out how to link said button to the G code I want it to send. but I don't want to take up too much of your time with this as i'm a complete novice. 

i've been using the macro buttons as examples, but their is multiple coding for each button throughout the file and so it's not as simple as I thought it would be. I was hoping to just replace something with "G90 X0 Y0 Z0" and have that button send it lol

winder, 2015-09-21:
The backend object has a command to send a raw command.

There is an abstraction layer so that different controllers can use different commands if they need to. For instance GRBL uses different commands for some features based on the version being used. Unfortunately this also adds more complexity to the code.

Jamieip, 2015-09-21:
How would I phrase the backend command for sending a raw command?

winder, 2015-09-21:
backend.sendGcodeCommand(cmd);

Jamieip, 2015-09-21:
Where would I place the command in that? Sorry about this haha

winder, 2015-09-21:
look at the "private void executeCustomGcode(String str)" function for an example

Jamieip, 2015-09-21:
Got it, thanks dude.

closed 2017-08-29 by winder