winder/Universal-G-Code-Sender #295: Work position and machine position coordinates do not update

KarlHeinz57, 2015-09-22:
UGS communicates with my Arduino UNO R3 board running GRBL fine. My laser cutter is not yet built but I'm exercising the Arduino/GRBL interface. During the first test the work position and machine position coordinates in the machine status updated as I exercised machine control. However, now they remain at zero. I've re-flashed GBRL and downloaded multiple versions of UGS and still the same behavior remains. By enabling verbose output I can see that the Arduino is indicating that the coordinates are being updated per any machine control initiated. The active state is seen to change to Running when machine control is initiated but the coordinates remain unchanged. I'm using the 2.0 UGS build (classic UI). I've also tried the 1.0.8 release with the same results. The GBRL version is 0.9j.20150811.

winder, 2015-09-23:
What happened between it working and not working?

KarlHeinz57, 2015-09-23:
Nothing in particular.  I thought perhaps I’d changed something critical with GRBL (played with $10 for instance).  Hence, I re-flashed the Arduino in the hopes of getting back to ground zero.

I considered trying the latest nightly build but the build status on GitHub indicates it failed.  The ZIP file dates are 9/13/15.

I did download the UGS source into NetBeans.  I had to make one correction(?) by removing “this.” from the following lines in MainWindow.java.

                settings.setCustomGcode1(this.customGcodeText1.getText());
                settings.setCustomGcode2(this.customGcodeText2.getText());
                settings.setCustomGcode3(this.customGcodeText3.getText());
                settings.setCustomGcode4(this.customGcodeText4.getText());
                settings.setCustomGcode5(this.customGcodeText5.getText());

Once compiled and run, the machine and work positions updated properly.

winder, 2015-09-23:
Thanks for pointing that out, I just made the fix. Does the nightly build work now?

KarlHeinz57, 2015-09-23:
Yes.  The work position coordinates update now.  What drives the machine position coordinates and why do they remain at zero? Thanks.

winder, 2015-09-23:
Aah, I just noticed that you said you were playing with $10 that is what caused the problem.

The status line you see when enabling verbose mode shows all the data UGS is getting, if you turn off WPos or MPos then UGS wont get the coordinates anymore.

closed 2015-09-23 by winder