winder/Universal-G-Code-Sender #297: Support for Tinyg G2 Firmware on Arduino Due

Mgilbride, 2015-09-24:
Am trying to get a sense of what you have supported in the current 2.0 nightly builds for the TinyG G2 on a Due? I can connect, query settings, and do some config. But I am having trouble getting settings to persist and can't seem to use text input mode rather than json input mode. Can you briefly outline what should and shouldn't work.

winder, 2015-09-24:
UGS wraps all the outgoing commands in JSON, so any manual commands are probably being wrapped in an unexpected way.

I tested connecting and  sending a file, but didn't go very far into configuration. In the future having a TinyG specific settings page similar to the Grbl settings page would make sense. I'm not sure how I should be handling manual commands though.

Mgilbride, 2015-09-24:
A quick way to get the feel is to connect through the arduino terminal. I
was explicitly writing json commands. But I like the grbl like / based text
commands better. Especially in firmware with so many settings!! I would
just turn json wrapping off whenever manual command is sent and let user
explicitly type json or text formatted commands. I think humans will prefer
text commands and robots don't really care. TinyG developers prefer json.

Mgilbride, 2015-09-24:
Also, settings page will need to be very selective. There are too many to
deal with without grouping, tabs, or something.

Mgilbride, 2015-10-01:
Ok. Almost a week of playing with tinyg and now understand a bit more of what's  going on. Lack of settings persistence is due to lack of eeprom. Do you mind pointing me to code that handles json command wrapping and reconfigure of gui  based on firmware. Hope to play around with it a bit.

winder, 2015-10-01:
The conversion happens in [TinyGGcodeCommand.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/types/TinyGGcodeCommand.java) and it's a sledgehammer approach for sure.

I think a better approach might be to use the no-frills "GCodeCommand" class in TinyGController and manage the JSON wrapping in a more explicit way, either in the Controller or Communicator objects. There needs to be some minor JSON parsing in [TinyGCommunicator.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/TinyGCommunicator.java
) to verify the 'ok' responses.

The API is probably going to need an extra hook to handle this.

Other than that [TinyGController.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/TinyGController.java) is a (mostly) blank canvas that needs to be fleshed out.

winder, 2015-10-01:
Also, com.google.gson is available for JSON parsing. It's used near the end of TinyGController to parse out TinyG responses.

winder, 2015-10-01:
For the GUI code you'll just need to make a new Window class in the NetBeans GUI builder (or however you want to build a Swing GUI).

Take a look at how [GrblFirmwareSettingsDialog.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/uielements/GrblFirmwareSettingsDialog.java) is handled for an example. I think I even used the GUI builder for managing the menu option, so there just needs to be a function to initialize the settings GUI with the "backend" object.

closed 2016-01-12 by winder