winder/Universal-G-Code-Sender #300: UGS Window Not Updating

barnold96, 2015-10-08:
I ran a job this morning while I was doing some yard work.  When I went to check after it finished, this is what I saw:

[image]

The cut had finished as expected but the UGS window didn't update properly.

winder, 2016-01-22:
There were a handful of weird edge conditions that could cause things like this. Over the past few months I've been cleaning up a lot of code which greatly simplified things to the point where I think these cases should be resolved now. These changes are in the nightly build.

How did the t-rex turn out?

closed 2016-01-22 by winder

barnold96, 2016-01-23:
Weird things?  Software?  Say it isn't so. lol

T-Rex is doing well, thinks!

--------------------------------------------

 

 
 
 it on GitHub.