winder/Universal-G-Code-Sender #302: Huge file support - UGS to use constant memory!

winder, 2015-10-14:
Many people have noticed that UGS doesn't support very large files. This is due to UGS storing several copies of the entire file in memory for the duration of an operation. Specifically there are 3 places that are storing the entire file in memory (with additional storage for metadata!):

1) Each line is stored in a GcodeCommand object, which has some additional metadata.
2) The visualizer is in use each vertex is stored in memory - for arcs this vertex data can be even larger than a single command..
3) The GUI even keeps its own copy as part of a Swing JTable component (GcodeTable.java), which stores even more metadata!!
4) (sigh) The GUI console may also keep a copy of this data, which would continue to grow as the operation is executed causing the application performance to degrade even further.

Basically a solution would need to address each of the above items to work in constant memory:

1) When commands are preprocessed they are built up into a collection of GcodeCommand objects. This could just as easily be piped into a new gcode file with the preprocessed data.

2) The visualizer is extremely naive in its implementation. I'm not sure how it could work in constant memory, but the memory usage could be optimized to do things like merge together tiny arc segments or line segments.

3) The data used by the JTable could be written to disk and looked up as needed. I'm sure there are a lot of options for this. One [possible resource](http://stackoverflow.com/questions/934927/efficient-way-to-implement-paged-sql-result-table-in-a-java-desktop-application), and [another](http://www.coderanch.com/t/345383/GUI/java/JTable-Paging).

4) The console output can be logged to a file, at which point there is probably a way to make the JTextArea discard old messages. A quick google turned up [this](http://compgroups.net/comp.lang.java.gui/buffered-jtextarea-or-equivalent/332460) which might be a solution.

winder, 2015-10-14:
Some issues this addresses #301, #244, #168

winder, 2015-12-25:
Possible solution for a disk backed JTable: [PersistentTableModel](https://github.com/Elblonko/kepler/blob/master/util/src/util/PersistentTableModel.java)

rugbymauri, 2015-12-26:
Is there really a need for the JTable? 
logging to the console, would be enough.

Or even the visualizer? I never need the visulizer.

winder, 2015-12-26:
The one nice thing you get from the jtable is that it matches up the
response with the command. But still I was thinking the same thing.

For the visualizer a lot of people use it as a preview tool. Maybe I'll
just add a warning when visualizing large files.

philreindl, 2016-01-10:
For what it's worth, I wouldn't miss the table.  IMO, the machine status indicator is sufficient to know whether it's keeping the buffer full.  

I was looking at things in the profiler (against master, not this branch), and it looks to me like a major issue is going to be the scrolling log.  We need to ensure that guy gets trimmed, and ideally doesn't create new objects constantly while scrolling. When run my machine with a laser, the GC pauses are significant.

winder, 2016-01-10:
@philreindl That was my conclusion as well, a stock jTable really isn't designed to perform well with this much data. I tried all sorts of things, but ended up leaving the table as-is but with a checkbox to disable it all together (and have it disabled by default).

In the constant memory branch I've already got the text log so that it wont grow over a certain amount, but objects are still getting allocated. Actually with a lot of the changes I'm making more objects are being allocated since I'm not keeping everything in memory anymore. Once I've got the memory issues reeled in I'll probably have to make another pass on the streaming performance.

closed 2016-01-18 by winder