winder/Universal-G-Code-Sender #303: Is the Jar file missing

rfgsoftware, 2015-10-17:
Is the Jar file missing from the latest release?  I can see it in previous releases, but not in 1..0.9

winder, 2015-10-18:
No, its in there. Click the link under Downloads.

closed 2015-10-18 by winder

rfgsoftware, 2015-10-18:
Really sorry, but I can't find it in the release files.  Where should I look?

winder, 2015-10-18:
Scroll down on the main github page, next to the screenshots

markstock, 2015-11-07:
Wow, you *really* need to make it clear that users should look in the "Downloads" section and not in the "Releases" or "Download .zip" or "git clone" buttons/features. I could have saved over an hour of my life, and some time from yours as well. I had no idea that the links under "Downloads" were any different than the bundles that github automatically makes.

winder, 2015-11-09:
@markstock I'm open to suggestions, I agree with you that there's a problem but I'm not sure how to improve the situation.

markstock, 2015-11-09:
@winder I'd be happy to help where I can. Let me work on the README.md and then I'll send you a patch (or I could branch and do the "pull request" thing).

winder, 2015-11-09:
@markstock Thanks Mark, whichever you prefer is fine by me. The only difference I can think of is that a pull request gets you listed as a project contributor which is important to some people.