winder/Universal-G-Code-Sender #306: Rasperry Pi, Raspian Jesse, and UGS 1.0.8 - clean new build not working.....

splitn2, 2015-11-03:
Hi, I just setup a brand new RPI with the Jesse image and did the updates, so its all up to date. I sftp'd UGS 1.0,8 onto it and unzipped it and changed permissions to 744 on the start.sh and UGS jar file. I've not been able to get it to start, and was wondering if there are any further steps? I'll try cut and paste the screen output later today. I got bCNC to run with ease, and I suspect some small issue is preventing UGS from running. I used the Jesse image off RaspberryPi.org, not NOOBs install I just flashed my SD with the latest Jesse .img file. It booted and does everything else just fine. Thanks, Richard

winder, 2015-11-03:
You should use the 2.0 nightly build. Other than that you just need a
recent version of Java installed, the rest of the dependencies are included
in the jar.

splitn2, 2015-11-03:
ahhh, sweet thanks!!!!! I just assumed that was bleeding edge so avoided it, will try tonight. cheers

splitn2, 2015-11-03:
Thanks @winder that sorted it!!!! I couldn't get the visualiser to work but the UGS otherwise worked perfectly!

closed 2015-11-06 by splitn2