winder/Universal-G-Code-Sender #308: Return to zero runaway

Erik46, 2015-11-05:
Overnight build 2.0 nov 5th 2015
At the end of a completed job I get a runaway when giving the command Return to Zero

**** Finished sending file. ****
// system was waiting here for the next instruction, I entered Return to Zero
ok
>>> G90 G0 Z396.0
>>> G90 G0 X0 Y0 Z0
ok

No idea where this 396 came from, my z axis only jas a useful 40mm before hitting the z axis limit switch, I hit the power off to motor button before this switch and stopped the machine without damage. I have seen it in an earlier issue where it was reported solved.

Erik46, 2015-11-09:
Situation improved with overnight build from nov 9th 2015
After reset zero I move a cubicle of +10 or +30 on ZXandY and then press return to zero. It first descends Z axis only to 8 and then all axes together to X0 Y0 and Z0 like the report said. The runaway seems gone. 

I do not know why this intermediate position is sent, To avoid obstacles I would do XY simultaneously and Z axis as last. I am already happy the runaway is gone.

ok
>>> G90 G0 Z7.999
>>> G90 G0 X0 Y0 Z0
ok

Erik46, 2015-11-11:
overnight build nov 11th 2015

The problem seems to be back again.

Grbl 0.9j ['$' for help]
>>> G21 G91 G0  Z10
ok
>>> G91 G0  Z10
ok
>>> G10 P0 L20 X0 Y0 Z0
ok
>>> G91 G0  Z10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Y10
ok
>>> G90 G0 Z30.0 // This line goves the problem
>>> G90 G0 X0 Y0 Z0
ok
ok

Erik46, 2015-11-11:
Correction, I have re-tested the issue with overnight build nov11th 2015

The line G90 G0 Z13.999 seems to do nothing, it descends in a straight line following the next instruction G90 G0 X0 Y0 Z0. This is correct behavior for the command to return to zero.

It would make sense to go to X0 Y0 first before descending Z in order to avoid as many obstacles, should the X0 Y0 be moved from the second line to the first?

**** Connected to COM3 @ 115200 baud ****

Grbl 0.9j ['$' for help]
>>> G21 G91 G0  Z14
ok
>>> G91 G0  X14
ok
>>> G91 G0  Y14
ok
>>> G90 G0 Z13.999
>>> G90 G0 X0 Y0 Z0
ok
ok

winder, 2015-11-11:
The return to zero command goes to the highest Z point UGS has seen, then goes to zero. I thought it went to X0Y0 before Z0 but it looks like thats not the case.

I think there might be a bug somewhere when units are changing between metric and imperial, but I haven't been able to reproduce it reliably.

Erik46, 2015-11-11:
Thanks for looking into this, I can spend some time in testing so if you can think of a test script I can do that for you.

 

Regards, erik

Erik46, 2015-11-25:
I have done some testing
UGS download 25nov2015 classic GUI, machine control in mm
Two times up a cube of 10mm and then descend 10mm, return to zero lifts the z axis to the highest point and then goes into a straight line to zero. Success. 
On a second attempt where I moved the x and y axis a bit further it went wrong again with a Z number way beyond my machine limits. 
On a third attempt I started UGS, changed to inches and step 0.4 hoping to replicate the second behavior. The response was different, it did not return to the highest point and went approx. 10mm too deep into the Z axis. 

A second observation, not necessarily connected to this issue is that my Z axis limit switches do not react while X and Y axis are ok, I have to investigate that.

Success:
>>> G91 G0  Z10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Z10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Z-10
ok
>>> G90 G0 Z19.999
>>> G90 G0 X0 Y0 Z0
ok
ok

Second attempt:
>>> G91 G0  Z10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X-10
ok
>>> G91 G0  Z10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X-10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X-10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X-10
ok
>>> G91 G0  Z-10
ok
>>> G90 G0 Z527.9735999999999
>>> G90 G0 X0 Y0 Z0
ok
ok

Third attempt in inches, step 0.4
ok
>>> G91 G0  X0.4
ok
>>> G91 G0  Y0.4
ok
>>> G91 G0  Z0.4
ok
>>> G91 G0  X0.4
ok
>>> G91 G0  Y0.4
ok
>>> G91 G0  X0.4
ok
>>> G91 G0  Y0.4
ok
>>> G91 G0  X0.4
ok
>>> G91 G0  Y0.4
ok
>>> G91 G0  Z-0.4
ok
>>> G90 G0 Z0.7697348484848486
>>> G90 G0 X0 Y0 Z0
ok
ok

Erik46, 2015-11-25:
Limit switches is Grbl, not UGS

Erik46, 2015-11-25:
I have done some more testing and went back to an earlier version.

Same path as earlier.

It seems that UGS 1.0.8. does it well, I have not seen any of these problems. When giving the return to zero it goes up to the highest point, then moves in x and y towards the zero and descends when arrived above zero. The difference seems in the G90 G28 X0 Y0 command at the end.

I am running against GRBL 0,8c now as I do need my limit switches to work for this type of testing

ok
>>> G91 G0  Z10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  Z10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  X10
ok
>>> G91 G0  Y10
ok
>>> G91 G0  Z-10
ok
>>> G21 G90 G0 Z19.999
>>> G90 G28 X0 Y0
ok
ok

nvisible, 2016-01-04:
I'm having the same issue with the Z-axis. If I jog the Z to +1mm and press "Return to Zero" the axis will run indefinitely in the positive direction. A full power cycle is the only way to reset.

jahnj0584, 2016-07-05:
This bug is still present and really annoying. It went as far as 600mm above the homed position, and broke one of my Z axis wires after it went past the limit switch! Can you change it so return to 0 makes it just go to the machine 0?

Erik46, 2016-07-05:
I have not had any problems on multiple projects running on GRBL v0.8c and Gcode sender v1.08 if you pay attention to:

-          After connecting the z-axis starts at 10.16 you have to reset zero

-          At the end of a run the z axis does not come back to zero so you need to reset zero before the next job

-          Default is inches so manual instructions move a lot if you forget to set this to mm

 

Your hard limit switches may be disabled due to a pinout change of the z-limit switch in GRBL between v0.8 and v0.9 

My hardware supports v0.8 so I am unable to do some re-testing now on v0.9

 

This issue was closed, another issue #433 was opened and looks to be the same. I suggest you add a comment to the open issue with reference to which version you refer to. (Windows – Java – Gcode sender (with date if 2.0) and GRBL)

 

Regards, erik

closed 2016-07-31 by winder