winder/Universal-G-Code-Sender #309: What .jar file?

markstock, 2015-11-06:
I thought I was a smart guy. The README says to double-click on "the jar file." Well, I unzipped the latest release, and git clone'd the repo, and there's no jar file that seems like it's obviously the one. What am I doing wrong? (Linux: Fedora/Ubuntu)

winder, 2015-11-06:
Read the readme. I wish I could remove the source download link github
automatically creates but I can't!!

markstock, 2015-11-07:
Thanks for the quick feedback. I've built tons of software from github, and if a compilation stage is required, it's always mentioned at the top of the README. I ignore the "Development" section if I'm not actually developing code for the project. So please forgive me for not reading the entire README. Can you please mention near the top of the README to look in the "Development" section for instructions on building the .jar file?

closed 2015-11-07 by markstock

markstock, 2015-11-07:
I realized that *only* the link in the "Download" section of the README has the .jar, as is described at the top of the README. I was off on a wild goose chase. I am used to github projects' READMEs being a big more clear about the installation instructions for new users, and was misled by differences between this and my other experiences. Things are working for me now and I am totally happy with the software.