winder/Universal-G-Code-Sender #310: Only some Arduino serial ports found on OSX

georgeHayRiv, 2015-11-06:
I am running OSX 10.11.1
I have tried the OSG and ClassicGUI versions with the same results.
An Arduino Leonardo shows up in the ports list but a Uno clone that uses the CH340 chip does not.
The clone is recognized by the Arduino IDE and the serial port exists when the following command is run in the terminal
ls /dev/tty*
/dev/tty.usbmodem621 is the Leonardo
/dev/tty.wchusbserial410 is the Uno clone

Here is the output from the command
/dev/tty	                                       /dev/ttysd

/dev/tty.Bluetooth-Incoming-Port	/dev/ttyse

/dev/tty.Bluetooth-Modem	       /dev/ttysf

/dev/tty.usbmodem621	             /dev/ttyt0

/dev/tty.wchusbserial410         /dev/ttyt1

/dev/ttyp0	                              /dev/ttyt2

The CH340 chipset does not appear to have a unique address like the other Arduinos. If one of these is plugged in, disconnected and another plugged in the same port the Arduino IDE sees it as the same device. Each real UNO needs to be picked in the Arduino IDE when it is plugged in.

AndrewRPerkins, 2016-01-05:
The CH340 chip not appearing is due to the way the program searches for serial ports, and is explained and resolved in this comment:

https://github.com/winder/Universal-G-Code-Sender/issues/261#issuecomment-122441168

winder, 2016-01-19:
Thanks @AndrewRPerkins.

As a workaround you can type /dev/tty.wchusbserial410 into the port field.

closed 2016-01-19 by winder

a-x-, 2019-01-12:
jfi, ⌘V does not work on macOS, but ^V does!
(who wants type it manually, huh??)