winder/Universal-G-Code-Sender #312: When I set the machine control to mm, then using Y100F100 my machine will move 100 mm @ 100 Inches per minute.

wlaw, 2015-11-14:


winder, 2016-01-12:
I don't change the feed rates.

closed 2016-01-12 by winder

wlaw, 2016-01-12:
Nice short answer,

Don't you think if I am working in mm the speed should be set to mm.

This dont seem correct or might cause a crash.

winder, 2016-01-12:
AFAIK G0 commands uses your GRBL settings, you should probably reduce your maximum feed rate settings if this is causing a problem.

Also, since I don't generate any feed rates with UGS, if this is whats happening it might be an issue with GRBL.

chamnit, 2016-01-12:
FYI, Grbl automatically converts all feed rate values to mm/min internally. This keeps this value consistent no matter what the user does in terms of setting their feed rate F value. Regardless of G20 inches mode or not.

wlaw, 2016-01-13:
All I can say is this. When the sender is in metric. And I manually, via
command tell my machine to move. Most of the settings are correct, the feed
rate stays in inches.

FYI, Grbl automatically converts all feed rate values to mm/min internally.
This keeps this value consistent no matter what the user does in terms of
setting their feed rate F value. Regardless of G20 inches mode or not.