winder/Universal-G-Code-Sender #314: Inches setting does not work unless I switch to mm then jog machine then switch back to inches.

jducen, 2015-11-23:
I discovered this issue using G38.2. I would invoke G38.2 Z-.05 F1 but UGS would return a probe failure after traveling a tiny distance. At first I thought it was a noise issue so I hooked up my Fluke and trended the probe but did not see anything out of the norm. Then I noticed that even though I had the units set to inches jogging in any direction would only move the machine a fraction of the 1" setting. Then I noticed the values shown in both machine position & work position would report having moved 1" even though actual distance traveled was no where close. I switched from inches to mm and jogged the machine then switched back to inches then everything seemed to work properly, including G38.2.

mza1979m, 2015-12-06:
Yes, I've been having the exact same issue since day one. I think it's a bug. Happens every day to me.

jducen, 2015-12-06:
Nice to know I'm not alone. I broke a bit figuring this out.

gregoryperrin, 2015-12-07:
I have exactly the same problem.

winder, 2015-12-07:
Thanks for the feedback everyone. I've been working on a big UGS optimization for the past few weeks, this is going to be the first thing I look at after thats complete.

gregoryperrin, 2015-12-07:
If I open UGCS and send a file made in inches (whith a g20 line at the begining), UGCS ignores it and acts like they are in mm. I need to jog manually in inches (ticking the "inches" box is not enough, i have to see it moving in inches), then send the gcode file, and then it works.

barnold96, 2015-12-26:
I don't have the G20 issue described above.  I modified the post processor in VCarve and PhotoVCarve to include a line with only G20 on it.  I had run into an issue with different software a few months ago in which it had a command group that included G20 and other commands, but G20 was not recognized.  Removing several of the commands that were not pertinent to my machine, G20 worked.

winder, 2016-02-20:
Sorry I misunderstood the bug report. I believe the latest version of GRBL will skip the entire line when there is an error. So if your generating gcode which puts G20 on the same line as something unsupported then your gcode fails to set the units properly.

UGS only changes the units to inch/mm for jogging. If you want to set units to inches, you can do that using the GRBL startup block: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#nxline---save-startup-block

closed 2016-02-20 by winder